AlexandreRouma/SDRPlusPlus #1516: [Windows 11] SDR++ v1.2.0 Crashes After Initialization Despite Antivirus Exclusion

KingAlAmir, 2024-11-07:
---

# BUG REPORT

**Hardware**
- **CPU**: Intel Core i7-8750H running at 4GHz, VT - Enabled
- **RAM**: Timetec 32GB kit DDR4 2667 MT/s
- **GPU**: NVIDIA GeForce GTX 1660 Ti
- **SDR**: RTL-SDR Blog V4 R828D RTL2832U

**Software**
- **Operating System**: Windows 11 Pro, Version 22H2, OS Build 22621.4317, Windows Feature Experience Pack 1000.22700.1041.0
- **SDR++**: Nightly Version 1.2.0 (latest), Installed on ‎01/12/2023

**Bug Description**
Upon launching SDR++, the program starts initializing and then abruptly closes without any error messages. Before closing, it creates several files, indicating some progress in loading modules. The application fails regardless of attempts to run with administrative privileges or through various command-line environments. Antivirus interference was ruled out by disabling Bitdefender and adding SDR++ to exclusions.

**Steps To Reproduce**
1. Double-clicking the `sdrpp.exe` file or running as administrator.
2. Running the executable through Command Prompt.
3. Running the executable through PowerShell.

**CMD Log Output**
```
C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64>.\sdrpp.exe -c
[07/11/2024 12:33:16.000] [INFO] SDR++ v1.2.0
[07/11/2024 12:33:16.000] [INFO] Loading config
[07/11/2024 12:33:16.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Using OpenGL 3.0
[07/11/2024 12:33:17.000] [INFO] Loading icons
[07/11/2024 12:33:17.000] [INFO] Loading band plans
[07/11/2024 12:33:17.000] [INFO] Loading band plans color table
[07/11/2024 12:33:17.000] [ERROR] Menu element is missing name key
[07/11/2024 12:33:17.000] [INFO] Loading modules
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/airspyhf_source.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\airspyhf_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/airspy_source.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\airspy_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/audio_sink.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\audio_sink_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/audio_source.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\audio_source_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/bladerf_source.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\bladerf_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/discord_integration.dll
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/file_source.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\file_source_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/fobossdr_source.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\fobossdr_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/frequency_manager.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\frequency_manager_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/hackrf_source.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\hackrf_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/hermes_source.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\hermes_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/iq_exporter.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\iq_exporter_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/limesdr_source.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\limesdr_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/m17_decoder.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\m17_decoder_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/meteor_demodulator.dll
[07/11/2024 12:33:17.000] [WARN] Recordings directory does not exist, creating it
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\meteor_demodulator_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/network_sink.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\network_sink_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/network_source.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\network_source_config.json' does not exist, creating it
[07/11/2024 12:33:17.000] [INFO] Loading C:/Users/Amirali/Downloads/sdrpp_windows_x64/sdrpp_windows_x64/modules/perseus_source.dll
[07/11/2024 12:33:17.000] [WARN] Config file 'C:\Users\Amirali\Downloads\sdrpp_windows_x64\sdrpp_windows_x64\perseus_config.json' does not exist, creating it
[... 135 lines cut ...]
perseus: Found device with VID/PID 5986:211B on BUS2 ADDR5
perseus: Found device with VID/PID 0BDA:2838 on BUS2 ADDR3
perseus: Found device with VID/PID 1D6B:0001 on BUS1 ADDR0
perseus: Found device with VID/PID 1532:0226 on BUS2 ADDR2
perseus: Found device with VID/PID 0BDA:0129 on BUS2 ADDR4
perseus: Found device with VID/PID 10DE:1AEC on BUS3 ADDR0
[07/11/2024 12:59:12.000] [INFO] Initializing PlutoSDR Source (plutosdr_source)
[07/11/2024 12:59:12.000] [INFO] Initializing RFNM Source (rfnm_source)
[07/11/2024 12:59:12.000] [INFO] Initializing RFspace Source (rfspace_source)
[07/11/2024 12:59:12.000] [INFO] Initializing RTL-SDR Source (rtl_sdr_source)
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
[07/11/2024 12:59:12.000] [INFO] Initializing RTL-TCP Source (rtl_tcp_source)
[07/11/2024 12:59:12.000] [INFO] Initializing Radio (radio)
[Resamp] predec: 32, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
...
```

**Additional Notes**
- Graphics Card: NVIDIA GeForce GTX 1660 Ti (GPU 2), Driver Version 32.0.15.6603 (Updated on 15/10/2024), DirectX Version 12 (FL 12.1)
- Enabled but not actively used: Intel(R) UHD Graphics 630 (GPU 0), Driver Version 26.20.100.6911 (Updated on 28/05/2019), DirectX Version 12 (FL 12.1)
- Bitdefender antivirus has been disabled and SDR++ is added to the exclusion list; issue persists.

---

KingAlAmir, 2024-11-07:
I also tested this with SDR++ v1.0.4, and I encountered the same issue. This suggests it may not be isolated to v1.2.0, as the crash happens in both versions despite setting up an antivirus exclusion. Please let me know if there are additional logs or specific steps I can take to assist with troubleshooting.

AlexandreRouma, 2024-11-07:
there's an issue with your audio setup. it crashes when trying to open an audio stream.

KingAlAmir, 2024-11-07:
I managed to resolve the issue, and here’s how I went about it:

After SDR++ kept force-closing during initialization, I downloaded and installed SDRangel (latest version), which automatically installed/updated the Microsoft Visual C++ 2015-2022 Redistributable (x64) - 14.40.33810. SDRangel worked fine after this installation, so I thought the issue might be related to a missing runtime.

However, SDR++ still failed to run and would crash during its initialization phase. Next, I downloaded SDRsharp (latest version), which required me to install the Microsoft Windows Desktop Runtime - 8.0.10 (x86) before it could run. After completing this installation, SDRsharp ran without any issues.

Following this, I tried running SDR++ one more time, and to my surprise, it successfully got past the initialization and didn’t crash. I still need to test all functionalities to ensure everything is working, but so far, SDR++ is running smoothly.

AlexandreRouma, 2024-11-07:
the redistributables are included with SDR++. I have a feeling you were missing media extensions in your windows installed. 
I also know for a fact that SDR++ runs fine on a clean windows 10 or 11 install.
Something's messed up about your specific install, and I can't tell you why. Just search the issues and you'll see that for the past 4 years nobody has had this exact issue so there must be something wrong.

closed 2024-11-07 by AlexandreRouma

KingAlAmir, 2024-11-07:
> the redistributables are included with SDR++. I have a feeling you were missing media extensions in your windows installed. I also know for a fact that SDR++ runs fine on a clean windows 10 or 11 [...]

--- 
Thanks for the insights! Even though updating the runtime seemed to help initially, I did a few more tests with audio connected and disconnected:

1. I retested by extracting SDR++ into a new folder and disconnected my headset (so no audio device was connected). In this setup, SDR++ got through the initialization phase without any crashes.

Regarding the redistributables, I didn’t initially realize they were included with SDR++. I’ve never had this issue before, so I thought I’d share some details on my current setup, which includes all the installed Windows runtimes and Visual C++ libraries:

[image]

[image]

This should give a clearer view of my configuration. I'll keep investigating to see if anything is specifically unusual in my setup, and I’ll report back if I find any other helpful details.

---