AlexandreRouma/SDRPlusPlus #1519: Feature: don't automatically reset config file when there is a problem

kord1e, 2024-11-07:
**Feature description**

Resetting config files automatically is very annoying when it happens. The user should be given a chance to fix them manually. A selection dialog (reset vs. quit and fix) might be useful.

AlexandreRouma, 2024-11-07:
No, it used to just quit when the config was corrupted, and the support load that caused was insufferable.
As long as you don't mess up a manual edit, it will never just wipe your config. And if it was corrupted due to a crash, you usually won't be able to recover it by hand anyway.

closed 2024-11-07 by AlexandreRouma

kord1e, 2024-11-24:
@AlexandreRouma A dialog would be useful so the user could then just reset and continue.

 In my case it happened because hard disk got full. Luckily only master config was corrupted and my freq bookmarks did not. Better make backups then.