AlexandreRouma/SDRPlusPlus #1520: More colors for bandplans

kord1e, 2024-11-07:
**Feature description**

I think it would be useful for custom bandplan makers if there were more colors (categories) in the default config, for example

Government/authorities 
Personal radio services (e.g. CB, PMR446, FRS)
Commercial users
Trunked radio networks
ISM and SRD/LPD433/part 15 equipment

and others if needed.

AlexandreRouma, 2024-11-07:
Has always been possible, you just need to add color definitions in the config file.
[image]

closed 2024-11-07 by AlexandreRouma

kord1e, 2024-11-24:
Anyone who makes new bandplan would benefit from it. Compatibility with SDR# is nice yes but extra colors would not harm that.