AlexandreRouma/SDRPlusPlus #1523: Rigctl client - Normal tuning mode option

qbg, 2024-11-09:
I want to use the rigctl_client module with my radio, but I use an SDR splitter/switch external to my rig rather than an IF tap. I want to see the signals as you normally do, but have rigctl_client sync the frequency of the demodulator to the rig.

f4bpp, 2025-05-08:
Hello. Don't hope any work about your request. I have same and asked for since more than a year. Like you and many others users, we want to synchronize the SDR++ frequency with our radio.

I'm using ChatGPT to make a program using Rigctl server to grab the frequency thanks to the rigctl server of sdr++ then send it with rigctld to my radio. It works but just some minutes because of rigctl server switch to idle state and the only way to restart it is to close sdr++ and run it again...

I'm desperate...

David.

AlexandreRouma, 2025-05-08:
> Don't hope any work about your request. I have the same request and asked Alexandre for that since more than a year

I'm not your personal software developer. If you want something quickly, write it yourself, or hire someone to write it for you.
You should consider yourself lucky that you're even getting 5 years of my work for free.
Entitled people like you are why so many open source developers give up on their projects.

f4bpp, 2025-05-08:
Please don't misinterpret. I understand the developers' priorities and that not all our requests can be met. I awkwardly expressed the fact that this feature was never your priority, which as a developer (far from possessing your talent and skills), I can understand. I'd like to remind you that I'm a teacher and that I had requested this feature for my educational projects, as I teach technology in two French secondary schools via educational projects linked to amateur radio. I'd also like to point out that I'd offered to pay you to add this functionality. So to say that because of people like me, developers of open source projects are giving up... don't exaggerate. 

Dear qgb, I'm developing an open source python program for my Yaesu FT-847 radio to do the job. I just have a problem with the figctl server function of SDR++ which deactivates itself after about 4 minutes without any intervention on the software or launching another application. I declared an issue, but out of anger, Alexandre deleted it. I hope that the problem will nevertheless be corrected and I'll gladly share my work so that all those who have been waiting for this feature can finally benefit from it because, let's not forget, under Linux, SDR++ is by far the best SDR software and I thank Alexandre again for developing it.

Peace and love, David.

f4bpp, 2025-05-09:
Dear Alexandre, 

I apologize for having raised a bug that wasn't a bug. As I said, I don't have your skills or talent to develop the feature I'm interested in, so I had to rely on artificial intelligence to help me. The latter was far from perfect and had produced some inappropriate code that caused rigctl to stall.

Unfortunately, when I read the issues, I realized that this problem had already been reported and I mistakenly thought that SDR++ had a bug, which is not the case.

I've now managed to code the functionality I'm so keen on, i.e. click on a frequency in SDR++'s bandscope and see it synchronized on my radio without having to use the rigctl client function and the IF frequency, which in my case needs to be adjusted on each of the amateur bands I work with.

I'll soon be making my entire code available on the Github platform after modifying it so that it's operational for all radios and not just mine as is currently the case. That way, if anyone wants to use it as an SDR++ integrated plugin rather than an add-on program as is currently the case, they'll be able to do so. I don't have the skills to do it myself, as I only have a basic knowledge of Python.

So I'd like to thank Alexandre for his work,  not having added this essential functionality for me, but for having produced a program that makes it possible to get with a little programming. I still offer to make a donation via Paypal to support all this amazing work. I just need a link.

Thanks a lot, David.

mflannery, 2026-03-19:
I worked with Claude AI and came up with a solution that I called rigctld_client (note the 'd' in the name).  This module will connect SDR++ to a running rigctld service that is connected to a radio (in my case its a Yaesu FT-100D).  With it, any change in frequency in SDR++ will immediately change the frequency on the radio.  This method works much better than the rigctl_server -> rigsync -> rigctld which would lag 5 to 10 second behind a change in frequency in SDR++.  With this new plugin, the frequency changes on the radio immediately. The frequency changes when the center frequency changes (sliding the frequencies on the screen) or when clicking in the spectrum or waterfall or directly adjusting the frequency.  I am very happy with this plugin.  

Unfortunately, I am not a git hero so I'm not sure how to add this module to the code for approval.  @AlexandreRouma , I am happy to either check in the code (with guidance from you on how to do that) or just paste the code here for you to review and check in.

Let me know how you'd like to proceed.
Mike