AlexandreRouma/SDRPlusPlus #1524: sdr++ cannot find rtl-sdr v4 as a source

linkwellfook, 2024-11-12:
**Hardware**
- CPU: AMD Ryzen 7 5825U
- RAM: 32 GB
- GPU: AMD Apu - Readon
- SDR: dongle - RTL-SDR Blog V4 R828D RTL2832U

**Software**
- Operating System: Ubuntu 24.04 VM
- SDR++: v1.2.1 nightly build (built at 18:31:07, Nov 9 2024)

**Bug Description**
The Source list does not include RTLSDRBlog Blog V4.  
I've followed the instructions as set in RTL-SDR blog and downloaded the most current nightly binary from sdr++ site.

I've test other sdr software and they find the dongle.

**Steps To Reproduce**
1. Follow steps RTL-SDR blog including blacklisting dvb_usb_rtl28xxu and dvb_usb_rtl2832u for good measure.  Shut down VM.  Plug dongle in, reboot run rtl_test.
2. Download the nightly build matching the Ubuntu build and use apt to install.
3. Launch SDR++, go to source.  RTLSDRBlog Blog V4 not listed

**Additional info**

The VM runs on top of Proxmox 8.2.7.  I have passed the usb through using the command 
qm set 206 -usb1 host=0bda:2838
In the vm when I use lsusb I see Realtek SemiConductor Corp. RTL2838 DV-T

AlexandreRouma, 2024-11-12:
Show the logs when running SDR++ in a terminal.

linkwellfook, 2024-11-12:
Hi

I've attached the output from the console.

[console_output.txt](https://github.com/user-attachments/files/17717593/console_output.txt)

Regards

AlexandreRouma, 2024-11-12:
Ok now show a screenshot of the source menu. The logs say the module is loaded so there must be a "RTL-SDR" option in the source selector.

linkwellfook, 2024-11-12:
Hi,

Attached is the screenshot of the Source dropdown and what is available.
I'm expecting to see RTL-SDR.
[image]

AlexandreRouma, 2024-11-12:
ok can you send your ~/.config/sdrpp/config.json file? Also what user are you running it as? SDR++ should never run as root.

linkwellfook, 2024-11-12:
Hi,
sure

config.json : 
[config.json](https://github.com/user-attachments/files/17717951/config.json)
I've not altered the config file.

the sdrpp app runs from a normal user account.  root is not used nor is sudo.

Regards

AlexandreRouma, 2024-11-12:
This is quite strange, please try this specific nightly build and see if the option shows up in the menu:
https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/11710871750

Don't reset any config, just install this over the existing version and see if something changes.

linkwellfook, 2024-11-12:
Hi

I downloaded the following file

sdrpp_ubuntu_noble_amd64.zip
extracted it.  The filename sdrpp_debian_amd64.deb

I installed it using apt install.
When I tried to run it I got a segment fault

[12/11/2024 14:55:19.000] [INFO] SDR++ v1.2.0
[12/11/2024 14:55:19.000] [INFO] Loading config
[12/11/2024 14:55:19.000] [INFO] Missing key in config decimationPower, repairing
[12/11/2024 14:55:19.000] [INFO] Missing key in config offset, repairing
[12/11/2024 14:55:19.000] [INFO] Missing key in config offsetMode, repairing
[12/11/2024 14:55:19.000] [INFO] Unused key in config decimation, repairing
[12/11/2024 14:55:19.000] [INFO] Unused key in config manualOffset, repairing
[12/11/2024 14:55:19.000] [WARN] ConfigManager locked, waiting...
Segmentation fault (core dumped)

The previous config folder and contents still remains in place.

Regards

AlexandreRouma, 2024-11-12:
ok yeah makes sense the config format was migrated. I guess renamed the config file to something like config.json.bak and let it generate its own. Won't give as much useful info but is at least something to try

linkwellfook, 2024-11-12:
Hi
renamed thje config.json file and ran sdrpp again.

No joy.  rtl-sdr is not listed in the Source dropdown.  
I'm included the regenerated config.json file
[config.json](https://github.com/user-attachments/files/17718492/config.json)

AlexandreRouma, 2024-11-12:
ok so good to know I didn't break anything in the 1.2.1 release.
Gonna be honest I have no idea how the logs you're getting are even possible. The init function of the module supposedly runs, but the post-init function doesn't, and the init function contains no branching so it not registering the source is suspicious...

linkwellfook, 2024-11-13:
Hi,

Just thought I would give a follow-up on this.

So I fired up a Ubuntu Jammy vm (Ubuntu 22.04.5 LTS) and installed sdr++ 1.2.1 built for Jammy.  RTL-SDR is listed as a source and it picked up the usb dongle.

I'm not sure if it is a compatibility issue with later versions of Ubuntu as later builds did move away from pulseaudio resulting in a change but sdrpp works in Jammy.

javierxio, 2024-11-19:
@AlexandreRouma Hello.

I want to also report the same issue. I am on latest Ubuntu 24.04 LTS and was not able to use SDR++ due to my dongle is not showing up.

Unfortunately, I am no able to test on an older version of Ubuntu (ex:  22.04 LTS).

So far, I had to try a different software [GQRX](http://gqrx.dk/) and I am able to use my RTL-SDK V4 USB dongle with no problem! 

(but I want to try SDR++ as it seems to have more features and a better UI than GQRX. Thanks.

AlexandreRouma, 2024-11-19:
I doubt it's the "same" issue. Show the logs.

javierxio, 2024-11-19:
@linkwellfook Hello. I thought I had the same issue as you, but the logs showed that I was missing some .so files; "No such file or directory"

```cmd
[18/11/2024 21:38:48.000] [INFO] SDR++ v1.2.1
[18/11/2024 21:38:48.000] [INFO] Loading config
[18/11/2024 21:38:48.000] [WARN] ConfigManager locked, waiting...
[18/11/2024 21:38:48.000] [INFO] Using OpenGL 3.0
[18/11/2024 21:38:48.000] [INFO] Loading icons
[18/11/2024 21:38:48.000] [INFO] Loading band plans
[18/11/2024 21:38:48.000] [INFO] Loading band plans color table
[18/11/2024 21:38:48.000] [INFO] Loading modules
[18/11/2024 21:38:48.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[18/11/2024 21:38:48.000] [INFO] Loading /usr/lib/sdrpp/plugins/iq_exporter.so
[18/11/2024 21:38:48.000] [INFO] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[18/11/2024 21:38:48.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/hackrf_source.so: libhackrf.so.0: cannot open shared object file: No such file or directory
[18/11/2024 21:38:48.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_client.so
[18/11/2024 21:38:48.000] [INFO] Loading /usr/lib/sdrpp/plugins/rfnm_source.so
[18/11/2024 21:38:48.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/rfnm_source.so: librfnm.so: cannot open shared object file: No such file or directory
[18/11/2024 21:38:48.000] [INFO] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so
[18/11/2024 21:38:48.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[18/11/2024 21:38:48.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so: libsdrplay_api.so.3: cannot open shared object file: No such file or directory
[18/11/2024 21:38:48.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[18/11/2024 21:38:48.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[18/11/2024 21:38:48.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/rtl_sdr_source.so: librtlsdr.so.2: cannot open shared object file: No such file or directory
```

I did some web search and found that I missed installing some dependencies.

Now, SDR++ is working on my Linux 24.04.

I can't find it now, but the dependencies are something like this:

https://github.com/AlexandreRouma/SDRPlusPlus/blob/11f87e0fe28153c3462941b194fae9a0b20ece1a/readme.md?plain=1#L180

@AlexandreRouma FYI.

amalgame21, 2024-12-02:
In android, sdr++ also cannot find rtl-sdr v4 when it is bebind a usb-c hub.

waltje, 2024-12-15:
For what it's worth... I just (12/15/2024) downloaded the nightly build (RPi 3B Bullseye ARMHF build) and, after installing a ton of armhf-based runtime, got it to work.  I do use my own updated build of the rtl-sdr library and tools (again, including for armhf..) and that works fine for me.

darkhelmet46, 2025-01-26:
I just installed the latest build of SDR++ 1.2.1 on Linux Mint Cinnamon and I used sdrpp_ubuntu_noble_amd64.deb because that is the version of Ubuntu that Cinnamon is based on.  It is not able to see my RTL-SDR V4.

waltje, 2025-01-26:
Make sure you have the *new* version (0.7.0 or up) of the RTL-SDR drivers (librtlsdr0 package) installed -olderones cannot connect/use the V4 modules.  The newer ones can, of course, also connect to the older ones.
--fred

darkhelmet46, 2025-01-26:
> Make sure you have the *new* version (0.7.0 or up) of the RTL-SDR drivers (librtlsdr0 package) installed -olderones cannot connect/use the V4 modules.  The newer ones can, of course, also connect to the older ones.
> --fred

Hmm, ok, I'm not sure exactly which version got installed.  I followed the instructions on their website:
https://www.rtl-sdr.com/rtl-sdr-quick-start-guide/

However, shortly after making my comment, I stumbled across this and now it works:
https://community.linuxmint.com/software/view/rtl-sdr

Unfortunately, I am one of those people who only knows just barely enough about Linux, etc,, to get himself into trouble.

closed 2025-02-21 by AlexandreRouma

seregayoga, 2025-10-17:
I tried to install from deb package  `sdrpp_ubuntu_noble_amd64.deb` and got the same problem. Similar logs as in https://github.com/AlexandreRouma/SDRPlusPlus/issues/1524#issuecomment-3094820696. Creating a link didn't help. But I built sdr++ from scratch and now it works. (Similar thread on Reddit https://www.reddit.com/r/RTLSDR/comments/1g2cdfe/sdr_not_seeing_rtlsdr_v4_in_the_source_menu_on/)

adisib, 2025-12-15:
If you don't want to build  SDR++ yourself, I just added `git reset --hard ab2434e` when building the rtl-sdr drivers to be before they changed the SONAME back to 0.