AlexandreRouma/SDRPlusPlus #1526: Apple blocks nightly SDR++ build for MacOS ARM

RobinCutshaw, 2024-11-13:
**Hardware**
- CPU: Apple M1 Ultra
- RAM: 64GB
- GPU: 
- SDR: RTL-SDR local

**Software**
- Operating System: MacOS Sonoma 14.6.1
- SDR++: Nightly 2024/11/09

Mac ARM nightly won't run.  Popup says: "SDR++” can’t be opened because Apple cannot check it for malicious software.

AlexandreRouma, 2024-11-13:
Not a bug, the app just isn't signed because I'm not paying apple hundreds of dollars.
[Apple itself tells you how to get around this.](https://support.apple.com/guide/mac-help/open-a-mac-app-from-an-unknown-developer-mh40616/mac)

closed 2024-11-13 by AlexandreRouma

RobinCutshaw, 2024-11-13:
I already pay them for the developer account.  Any interest in a signature?

Robin

Terminator0007, 2024-11-13:
Any problems to open nightly version for mac OS ARM. Only to accept opening in security section.

RobinCutshaw, 2024-11-13:
That works.  It is just nice to have trusted software.

AlexandreRouma, 2024-11-13:
I don't want to rely on someone else's keys, so this is how it will stay.

denisbondar, 2025-01-28:
> Any problems to open nightly version for mac OS ARM. Only to accept opening in security section.

I have macOS 15.2. Can you tell me how I can launch the application? The operating system offers only to remove the application.

johntdavis84, 2025-01-28:
Try this:

1. Try to open the app, dismiss the error.
2. Go to System Settings > Privacy and Security.
3. Scroll down until you see Security at the bottom.
4. There should be a message about it having blocked SDR++. There will be a button to open it anyway. You only hav to do this once.