AlexandreRouma/SDRPlusPlus #1528: Audio not working in Ubuntu 24.10 because SDR++ cannot load audio sink module

davidecorradini, 2024-11-15:
**Hardware**
- CPU: Intel® Core™ i7-6700HQ
- RAM: 16GB
- GPU: Nvidia GeForce GTX 960M
- SDR: RTL-SDR v4 (remote, I'm running a SDR++ server in another room)

**Software**
- Operating System: Ubuntu 24.10 (Kernel: Linux 6.11.0-8-generic)
- SDR++: 1.2.1 (Built on Nov 9, 2024)

**Bug Description**
I recently updated my system to Ubuntu 24.04, and SDR++ cannot load the audio sink module. In the *sinks* menu, I can only choose `None` and `Network` as sinks. In the terminal I get this error when SDR++ is starting:

```
[15/11/2024 09:27:21.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/audio_sink.so: librtaudio.so.6: cannot open shared object file: No such file or directory
```

However, the file `/usr/lib/sdrpp/plugins/audio_sink.so` exists.

The rest of the software runs properly. I can connect to the remote SDR and see the graphs. I just can't listen to the demodulated audio.

**Steps To Reproduce**
1. Start SDR++
2. Audio sink is not available

**Screenshots**
Not required.

**Additional info**
For further details, I'll post here the full startup lig of SDR++.
```
[15/11/2024 09:27:21.000] [INFO] SDR++ v1.2.1
[15/11/2024 09:27:21.000] [INFO] Loading config
[15/11/2024 09:27:21.000] [INFO] Using OpenGL 3.0
[15/11/2024 09:27:21.000] [ERROR] Glfw Error 65548: Wayland: The platform does not support setting the window icon
[15/11/2024 09:27:21.000] [INFO] Loading icons
[15/11/2024 09:27:21.000] [INFO] Loading band plans
[15/11/2024 09:27:21.000] [INFO] Loading band plans color table
[15/11/2024 09:27:21.000] [INFO] Loading modules
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/file_source.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/tetra_demodulator.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_source.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/vhfvoiceradio.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/recorder.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_source.so
[15/11/2024 09:27:21.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/audio_source.so: librtaudio.so.6: cannot open shared object file: No such file or directory
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[15/11/2024 09:27:21.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/rtl_sdr_source.so: librtlsdr.so.2: cannot open shared object file: No such file or directory
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[15/11/2024 09:27:21.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/audio_sink.so: librtaudio.so.6: cannot open shared object file: No such file or directory
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/m17_decoder.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_sink.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/rfnm_source.so
[15/11/2024 09:27:21.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/rfnm_source.so: librfnm.so: cannot open shared object file: No such file or directory
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/radio.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/bladerf_source.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[15/11/2024 09:27:21.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/plutosdr_source.so: libiio.so.0: cannot open shared object file: No such file or directory
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[15/11/2024 09:27:21.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so: libsdrplay_api.so.3: cannot open shared object file: No such file or directory
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/scanner.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[15/11/2024 09:27:21.000] [INFO] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so
[... 92 lines cut ...]
[Resamp] predec: 64, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Resamp] predec: 64, interp: 112, decim: 125, inacc: 0.000000%, taps: 9500
[15/11/2024 09:27:22.000] [INFO] SDRPPServerSourceModule 'SDR++ Server Source': Tune: 96175000.000000!
[15/11/2024 09:27:22.000] [INFO] Running post-init for Airspy Source
[15/11/2024 09:27:22.000] [INFO] Running post-init for AirspyHF+ Source
[15/11/2024 09:27:22.000] [INFO] Running post-init for BladeRF Source
[15/11/2024 09:27:22.000] [INFO] Running post-init for File Source
[15/11/2024 09:27:22.000] [INFO] Running post-init for Frequency Manager
[15/11/2024 09:27:22.000] [INFO] Running post-init for HackRF Source
[15/11/2024 09:27:22.000] [INFO] Running post-init for Hermes Source
[15/11/2024 09:27:22.000] [INFO] Running post-init for LimeSDR Source
[15/11/2024 09:27:22.000] [INFO] Running post-init for Network Sink
[15/11/2024 09:27:22.000] [INFO] Running post-init for RFspace Source
[15/11/2024 09:27:22.000] [INFO] Running post-init for RTL-TCP Source
[15/11/2024 09:27:22.000] [INFO] Running post-init for Radio
[15/11/2024 09:27:22.000] [INFO] Running post-init for Radio2
[15/11/2024 09:27:22.000] [INFO] Running post-init for Recorder
[15/11/2024 09:27:22.000] [INFO] Running post-init for Rigctl Server
[15/11/2024 09:27:22.000] [INFO] Running post-init for SDR++ Server Source
[15/11/2024 09:27:22.000] [INFO] Running post-init for Scanner
[15/11/2024 09:27:22.000] [INFO] Running post-init for SpyServer Source
[15/11/2024 09:27:22.000] [INFO] Running post-init for Tetra Decoder
[15/11/2024 09:27:22.000] [INFO] Running post-init for VHF Voice Radio
[15/11/2024 09:27:22.000] [INFO] Ready.
```

AlexandreRouma, 2024-11-15:
you tried to install a 24.04 deb on 24.10, not gonna work.

closed 2024-11-15 by AlexandreRouma

davidecorradini, 2024-11-15:
@AlexandreRouma is the 24.10 deb available for download? I don't see it. There are other ways to make it run on this Ubuntu version? Thanks

AlexandreRouma, 2024-11-15:
there's not, you have to build from source for now.
I don't use ubuntu, so I don't provide a build as soon as they release a new version. There will be one when I'll have time to configure it and check that everything works properly.