AlexandreRouma/SDRPlusPlus #1539: SDR++ crashes before fully open

chrisjosack, 2024-12-01:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Intel(R) Core(TM) i3-8130U CPU @ 2.20GHz   2.21 GHz
- RAM: 6.00 GB (5.88 GB usable)
- GPU: 
- SDR: Remote using Airspy Server V2.0.1700 on Linux Mint 20.3 Una 

**Software**
- Operating System: Windows 11 Home, Ver 23H2, 64-bit operating system, x64-based processor
- SDR++: V1.2.1, Version nightly update downloaded 11/30/2024

**Bug Description**
Starting SDR++ from the .exe file show only a brief banner then it closes.  Running SDRPP.exe -c from command line and it crashes at "[INFO] Initializing BladeRF Source (bladerf_source)".  BladeRF was never installed on this machine. SDRPP worked fine until several days ago.

**Steps To Reproduce**
1. ...E:\sdrpp_windows_x64>.\sdrpp.exe -c
[01/12/2024 13:35:05.000] [INFO] SDR++ v1.2.1
[01/12/2024 13:35:05.000] [INFO] Loading config
[01/12/2024 13:35:05.000] [INFO] Using OpenGL 3.0
[01/12/2024 13:35:05.000] [INFO] Loading icons
[01/12/2024 13:35:05.000] [INFO] Loading band plans
[01/12/2024 13:35:05.000] [INFO] Loading band plans color table
[01/12/2024 13:35:05.000] [INFO] Loading modules
[01/12/2024 13:35:05.000] [INFO] Loading E:/sdrpp_windows_x64/modules/airspyhf_source.dll
[01/12/2024 13:35:05.000] [INFO] Loading E:/sdrpp_windows_x64/modules/airspy_source.dll
[01/12/2024 13:35:05.000] [INFO] Loading E:/sdrpp_windows_x64/modules/audio_sink.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/audio_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/bladerf_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/discord_integration.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/file_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/fobossdr_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/frequency_manager.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/hackrf_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/hermes_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/iq_exporter.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/limesdr_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/m17_decoder.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/meteor_demodulator.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/network_sink.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/network_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/perseus_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/plutosdr_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/radio.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/recorder.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/rfnm_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/rfspace_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/rigctl_client.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/rigctl_server.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/rtl_sdr_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/rtl_tcp_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/scanner.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/sdrplay_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/sdrpp_server_source.dll
[01/12/2024 13:35:06.000] [INFO] Loading E:/sdrpp_windows_x64/modules/spyserver_source.dll
[01/12/2024 13:35:06.000] [INFO] Initializing Airspy Source (airspy_source)
[01/12/2024 13:35:06.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[01/12/2024 13:35:06.000] [INFO] Initializing Audio Sink (audio_sink)
[01/12/2024 13:35:06.000] [INFO] Initializing Audio Source (audio_source)
[01/12/2024 13:35:06.000] [INFO] Initializing BladeRF Source (bladerf_source)
E:\sdrpp_windows_x64>

2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

AlexandreRouma, 2024-12-01:
There likely is a problematic USB device connected to the computer. Unplug everything except your mouse and keyboard and see if SDR++ starts up correctly. If it does start correctly, plug devices back in one by one and see which one prevents SDR++ from starting.

In the case that this does not work, just delete the `modules/bladerf_source.dll` file since you do not appear to need it for your use code.

chrisjosack, 2024-12-04:
Dear Alexandre,
I unplugged the USB mouse and it failed at the same spot Initializing "BladeRF Source (bladerf_source)". Next I deleted the "modules/bladerf_source.dll" but it failed. I removed several other modules/SDR dll's. but it ultimately failed.

Thanks for your help
Chris

C:\Users\Chris\Desktop\sdrpp_windows_x64>sdrpp.exe -c
[04/12/2024 15:31:46.000] [INFO] SDR++ v1.2.1
[04/12/2024 15:31:46.000] [INFO] Loading config
[04/12/2024 15:31:47.000] [INFO] Using OpenGL 3.0
[04/12/2024 15:31:47.000] [INFO] Loading icons
[04/12/2024 15:31:47.000] [INFO] Loading band plans
[04/12/2024 15:31:47.000] [INFO] Loading band plans color table
[04/12/2024 15:31:47.000] [INFO] Loading modules
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/airspyhf_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/airspy_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/audio_sink.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/audio_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/discord_integration.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/file_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/frequency_manager.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/hermes_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/iq_exporter.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/limesdr_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/m17_decoder.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/meteor_demodulator.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/network_sink.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/network_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/perseus_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/plutosdr_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/radio.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/recorder.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/rfnm_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/rfspace_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/rigctl_client.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/rigctl_server.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/rtl_sdr_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/rtl_tcp_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/scanner.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/sdrplay_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/sdrpp_server_source.dll
[04/12/2024 15:31:47.000] [INFO] Loading C:/Users/Chris/Desktop/sdrpp_windows_x64/modules/spyserver_source.dll
[04/12/2024 15:31:47.000] [INFO] Initializing Airspy Source (airspy_source)
[04/12/2024 15:31:47.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[04/12/2024 15:31:47.000] [INFO] Initializing Audio Sink (audio_sink)
[04/12/2024 15:31:47.000] [INFO] Initializing Audio Source (audio_source)
[04/12/2024 15:31:47.000] [INFO] Initializing BladeRF Source (bladerf_source)
[04/12/2024 15:31:47.000] [**ERROR**] Module 'bladerf_source' doesn't exist
[04/12/2024 15:31:47.000] [INFO] Initializing File Source (file_source)
[04/12/2024 15:31:47.000] [INFO] Initializing FobosSDR Source (fobossdr_source)
[04/12/2024 15:31:47.000] **[ERROR**] Module 'fobossdr_source' doesn't exist
[04/12/2024 15:31:47.000] [INFO] Initializing Frequency Manager (frequency_manager)
[04/12/2024 15:31:47.000] [INFO] Initializing HackRF Source (hackrf_source)
[04/12/2024 15:31:47.000] [**ERROR**] Module 'hackrf_source' doesn't exist
[04/12/2024 15:31:47.000] [INFO] Initializing Harogic Source (harogic_source)
[04/12/2024 15:31:47.000] [**ERROR**] Module 'harogic_source' doesn't exist
[04/12/2024 15:31:47.000] [INFO] Initializing Hermes Source (hermes_source)
[04/12/2024 15:31:47.000] [INFO] Initializing LimeSDR Source (limesdr_source)
[04/12/2024 15:31:47.000] [INFO] Initializing Network Sink (network_sink)
[04/12/2024 15:31:47.000] [INFO] Initializing Network Source (network_source)
[04/12/2024 15:31:47.000] [INFO] Initializing PerseusSDR Source (perseus_source)
perseus: perseus_init()
perseus: EXE: [C:\Users\Chris\Desktop\sdrpp_windows_x64\sdrpp.exe]
perseus: path: [C:\Users\Chris\Desktop\sdrpp_windows_x64\] name: [sdrpp.exe]
libusb: error [enumerate_hcd_root_hub] program assertion failed - HCD 'ROOT\SYSTEM\0001' child not found

AlexandreRouma, 2025-01-01:
The libsub error just confirms there is some defective USB device in or connected to your computer, that's about all I can say. Since you're only gonna use spyserver, delete every source module other than the spyserver one and it'll work fine, but that still means that if you ever plan to use a real SDR it won't work properly with any software anyway.

closed 2025-01-01 by AlexandreRouma