AlexandreRouma/SDRPlusPlus #1540: Can't find the recorded file and can't change the location of the recorded file (clicking three dots on the right doesn't work)

lwlwlel, 2024-12-05:
**Hardware**
- CPU: Snapdragon 870
- RAM:12Gb
- GPU: 
- SDR: RTLSDR V4

**Software**
- Operating System: Android14
- SDR++: Ver 1.2.0 the build date I cannot find in Android system

**Bug Description**
I can't find the recorded file and I can't change the location of the recorded file (clicking three dots on the right doesn't work)

Uploading SVID_20241206_005534_1.mp4…

lwlwlel, 2024-12-05:
https://github.com/user-attachments/assets/e6015d52-47cc-40a5-8567-2d2cf7bef87f

Mraanderson, 2024-12-20:
Have you tried tapping into the address field and entering it manually as /**storage/emulated/0/music/** or whichever folder? %root%/recordings didn't work for me. It should change from Red to White if the path is good.
[image]

The browse button not working is listed in the readme...you did read the reame didn't you?

Mraanderson, 2024-12-20:
[image]

AlexandreRouma, 2025-02-21:
file selection dialogs aren't supported on android because of how the OS works. you'll have to go grab the file with ADB in the app files if you didn't change the path manually.

closed 2025-02-21 by AlexandreRouma