AlexandreRouma/SDRPlusPlus #1544: Debian Trixie release package is missing

vkomenda, 2024-12-09:
**Feature description**

There needs to be a package for Trixie which is scheduled for release in April 2025.

I wrote a fix https://github.com/AlexandreRouma/SDRPlusPlus/commit/ca61b98c926d70e43c985ec9193316ec3ce33f40

Let me know if you want to cherry-pick or for me to make a PR.

AlexandreRouma, 2024-12-09:
I don't provide package before a distro is released.

AlexandreRouma, 2024-12-09:
I only support sid because a friend of mine uses it otherwise I just wouldn't bother with it. I don't use debian, I don't want to waste my time with a version of a distro that is not finished.

vkomenda, 2024-12-09:
OK, fair enough. The build files for sid and trixie are identical. The base docker images are different though. So it's not quite the same thing.