AlexandreRouma/SDRPlusPlus #1548: VFO Lock button, and waterfall slide

cole8085, 2024-12-16:
I'd like to see a VFO and settings lock to avoid accidental frequency changes, or settings from being changed, while allowing something like a sat. tracking plugin to continue making adjustments. : And to be able to click > drag water fall to visually scan rather than jump +- ~SDR base band size.

AlexandreRouma, 2024-12-16:
duplicate of #1298

closed 2024-12-16 by AlexandreRouma

cole8085, 2024-12-18:
> duplicate of #1298

Sorry I thought I searched that one, did you put the click and drag of the waterfall on the to do list?

AlexandreRouma, 2024-12-18:
Already a thing, just that you have to grab the frequency scale and not the waterfall.

cole8085, 2024-12-19:
> Already a thing, just that you have to grab the frequency scale and not the waterfall.

Oh ok, I never tried that, thanks.