AlexandreRouma/SDRPlusPlus #1554: Tuned frequency not retained over restart, RTL-SDR at 1.024 MHz sampling

gm8arv, 2024-12-19:
[2024.12.19.command.output.txt](https://github.com/user-attachments/files/18193808/2024.12.19.command.output.txt)
[2024.12.19.logfile.txt](https://github.com/user-attachments/files/18193809/2024.12.19.logfile.txt)
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Intel i7-4770K quad-core
- RAM: 16GB
- GPU:  GT730
- SDR: (Remote or local? If remote, what protocol?) Local RTL-SDR V3

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15) WIn-10 x64
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software). SDR++ v1.2.1 (build 03:37:20, Dec 19, 2024)

**Bug Description**
Tuned frequency not restored on program restart

**Steps To Reproduce**
SDR++ v1.2.1 (build 03:37:20, Dec 19, 2024)
Source RTL-SDR
Sampling 1.024 MHz
Radio: WFM, 200000 Hz

At startup, shows 100.000.000 Hz, centre tuning selected
Tuned to 92.1 MHz using a Frequency Manager entry
Shows 92.100.000 Hz
Run, Stop and closedown (with top-right "x")
Restart program, shows 100.000.000 Hz.

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Not tested with other hardware or other sampling rates.
Logs and command output attached.

KubaPro010, 2024-12-19:
yeah, it happens with every single sdr because sdrpp just doesn't save the vfo frequencies, only center

gm8arv, 2024-12-19:
Thanks.  

May I suggest that saving the VFO frequency be added as an option?  I, and others, find it confusing that the program opens on a completely unrelated frequency.

AlexandreRouma, 2024-12-19:
> yeah, it happens with every single sdr because sdrpp just doesn't save the vfo frequencies, only center

That is 100% wrong.

AlexandreRouma, 2024-12-20:
bug in the source module initialization sequence causing the VFO offset correction to be done before the final source module has been loaded and initialized.