AlexandreRouma/SDRPlusPlus #1558: SDRPlay source not found on Macos build. (SDRPlay 3.15 api, Dec 28 2024)

edfardos, 2024-12-29:
**Hardware**
- CPU: Apple M4
- RAM:  512G
- GPU:  Apple M4
- SDR: Local SDRPlay

**Software**
- Operating System: MacOS Sequoia 15.1
- SDR++:  Nightly compiled 28/12/2024 18:34:32.000

**Bug Description**
SDRPlay is not available to select as a source

**Steps To Reproduce**
1. Install SDRPlay API v3.15, library observed in libsdrplay_api.so.3.152. 
2. Compile SDR++ with SDRPlay source ON, ./Plugins/sdrplay_source.dylib file observed
3. Create MacOS ".app", move to Applications, 
4. Start SDR++, SDRPlay is not an option to select as a source, not listed in modules.

craiger@Craigs-Mac-mini Plugins % pwd
/Volumes/Macintosh HD/Applications/SDR++.app/Contents/Plugin

craiger@Craigs-Mac-mini Plugins % ls -alh sdrplay*
-rwxr-xr-x  1 craiger  staff   141K Dec 28 15:52 sdrplay_source.dylib

/Volumes/Macintosh\ HD/Applications/SDR++.app/Contents/MacOS/sdrpp 
[28/12/2024 18:37:49.000] [INFO] SDR++ v1.2.1
[28/12/2024 18:37:49.000] [INFO] Loading config
[28/12/2024 18:37:49.000] [WARN] ConfigManager locked, waiting...
[28/12/2024 18:37:50.000] [ERROR] Glfw Error 65548: Cocoa: Regular windows do not have icons on macOS
[28/12/2024 18:37:50.000] [INFO] Loading icons
[28/12/2024 18:37:50.000] [INFO] Loading band plans
[28/12/2024 18:37:50.000] [INFO] Loading band plans color table
[28/12/2024 18:37:50.000] [INFO] Loading modules
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/libsdrplay_api.dylib
[28/12/2024 18:37:50.000] [ERROR] /Applications/SDR++.app/Contents/Resources/../Plugins/libsdrplay_api.dylib is missing _INFO_ symbol
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rtl_sdr_source.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/meteor_demodulator.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/airspy_source.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/network_sink.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/scanner.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib
[28/12/2024 18:37:50.000] [ERROR] Couldn't load /Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib: dlopen(/Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib, 0x0005): Library not loaded: @rpath/libsdrplay_api.so.3
  Referenced from: <5E93C732-81D8-3B6C-858E-28D9726A6E9B> /Applications/SDR++.app/Contents/Plugins/sdrplay_source.dylib
  Reason: tried: '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/Frameworks/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/MacOS/../Frameworks/libsdrplay_api.so.3' (no such file)
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/radio.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rigctl_server.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/audio_sink.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/frequency_manager.dylib
[28/12/2024 18:37:50.000] [WARN] ConfigManager locked, waiting...
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/recorder.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/file_source.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/network_source.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/bladerf_source.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/airspyhf_source.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/spyserver_source.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rigctl_client.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/new_portaudio_sink.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/hermes_source.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/discord_integration.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rfspace_source.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rtl_tcp_source.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/hackrf_source.dylib
[28/12/2024 18:37:50.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/sdrpp_server_source.dylib
[28/12/2024 18:37:50.000] [INFO] Initializing Airspy Source (airspy_source)
[28/12/2024 18:37:50.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[28/12/2024 18:37:50.000] [INFO] Initializing Audio Sink (audio_sink)
[28/12/2024 18:37:50.000] [INFO] Initializing Audio Source (audio_source)
[28/12/2024 18:37:50.000] [ERROR] Module 'audio_source' doesn't exist
[28/12/2024 18:37:50.000] [INFO] Initializing BladeRF Source (bladerf_source)
[... 60 lines cut ...]
[28/12/2024 18:37:50.000] [INFO] Audio device open.
[28/12/2024 18:37:50.000] [INFO] RTLSDRSourceModule 'RTL-SDR Source': Tune: 100000000.000000!
[28/12/2024 18:37:50.000] [INFO] Running post-init for Airspy Source
[28/12/2024 18:37:50.000] [INFO] Running post-init for AirspyHF+ Source
[28/12/2024 18:37:50.000] [INFO] Running post-init for Audio Sink
[28/12/2024 18:37:50.000] [INFO] Running post-init for BladeRF Source
[28/12/2024 18:37:50.000] [INFO] Running post-init for File Source
[28/12/2024 18:37:50.000] [INFO] Running post-init for Frequency Manager
[28/12/2024 18:37:50.000] [INFO] Running post-init for HackRF Source
[28/12/2024 18:37:50.000] [INFO] Running post-init for Hermes Source
[28/12/2024 18:37:50.000] [INFO] Running post-init for Network Sink
[28/12/2024 18:37:50.000] [INFO] Running post-init for Network Source
[28/12/2024 18:37:50.000] [INFO] Running post-init for RFspace Source
[28/12/2024 18:37:50.000] [INFO] Running post-init for RTL-SDR Source
[28/12/2024 18:37:50.000] [INFO] Running post-init for RTL-TCP Source
[28/12/2024 18:37:50.000] [INFO] Running post-init for Radio
[28/12/2024 18:37:50.000] [INFO] Running post-init for Recorder
[28/12/2024 18:37:50.000] [INFO] Running post-init for Rigctl Server
[28/12/2024 18:37:50.000] [INFO] Running post-init for SDR++ Server Source
[28/12/2024 18:37:50.000] [INFO] Running post-init for SpyServer Source
[28/12/2024 18:37:50.000] [INFO] Ready.
2024-12-28 18:37:50.506 sdrpp[6523:93387] +[IMKClient subclass]: chose IMKClient_Modern
[28/12/2024 18:38:06.000] [INFO] Exiting successfully

I'm a new mac person, please be gentle.   Linux build works great!

edfardos, 2024-12-29:
We're able to find the library after copying it to the expected location:
cp /Library/SDRplayAPI/3.15.0/lib/libsdrplay_api.so.3.15 ./libsdrplay_api.so.3

sdrpp now results in the following when you press "play" in sdrpp:
[28/12/2024 18:57:45.000] [ERROR] Could not select RSP device: sdrplay_api_Fail
[28/12/2024 18:57:45.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Tune: 100000000.000000!

edfardos, 2024-12-29:
% system_profiler SPUSBDataType

            Vendor-Specific Device:
              Product ID: 0x3050
              Vendor ID: 0x1df7
              Version: 2.00
              Speed: Up to 480 Mb/s
              Location ID: 0x02110000 / 3
              Current Available (mA): 500
              Extra Operating Current (mA): 0

edfardos, 2024-12-29:
... and SDRConnect, from SDRPlay works.  assuming it uses the same sdrplay_apiService.

edfardos, 2024-12-29:
cmake .. -DOPT_BUILD_SOAPY_SOURCE=OFF -DOPT_BUILD_BLADERF_SOURCE=OFF -DOPT_BUILD_AUDIO_SOURCE=OFF -DOPT_BUILD_AUDIO_SINK=OFF -DOPT_BUILD_PORTAUDIO_SINK=ON -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON -DOPT_BUILD_M17_DECODER=ON -DUSE_BUNDLE_DEFAULTS=ON -DCMAKE_BUILD_TYPE=Release -DOPT_BUILD_SDRPLAY_SOURCE=ON -DOPT_BUILD_PLUTOSDR_SOURCE=OFF -DOPT_BUILD_AIRSPY_SOURCE=OFF -DOPT_BUILD_AIRSPYHF_SOURCE=OFF -DOPT_BUILD_M17_DECODER=OFF

edfardos, 2024-12-29:
Downgrading SDRPlay API to 3.14 (after compilation steps above) results in a working sdrplay_apiService.  SDR++ starts, works, for about a minute, then "stops"  as sdrplay_apiService reports:

libusb: warning [darwin_transfer_status] transfer error: device not responding (value = 0xe00002ed)
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = In): pipe is stalled (code = 0xe000404f)
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = In): pipe is stalled (code = 0xe000404f)
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = In): pipe is stalled (code = 0xe000404f)
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = In): pipe is stalled (code = 0xe000404f)

.. tried a few usb cables and ports.

edfardos, 2024-12-29:
Downgrading SDRPlay API to v3.14 (from 3.15), after completing all above steps solves the problem.  Ticket filed with SDRPlay guys.

edfardos, 2024-12-30:
.. and after about 15 minutes the sdrplay_apiService starts spitting out USB watchdog errors, USB lockup, and sdr++ stops playing.

SDRPlay responded quickly to the ticket, indicating there was nothing wrong with the 3.15 api because SDRTrunc works.   I'm guessing they haven't tested macos15.1, where the api service requires a kill-9 to stop....

N1FKO, 2025-01-09:
FYI on all my Macs running macOS Sequoia (15.2), a clean install of API 3.15 fails to report RSP1Bs to SDRTrunk, SDRAngel, etc.  API 3.14 works.

I believe an upgrade from 3.14 to 3.15 appears to work because the old service is lingering around, but once you reboot it's broken until I uninstall and reinstall 3.14.

SDRConnect works because (reportedly) it does not use the API package, having all support built in.

Oddly enough, 3.15 does report my old RSP1.

AlexandreRouma, 2025-02-21:
the SDRplay API returning an error is out of the control of SDR++, nothing I can do about it.

And support for the RSP1B was added around month ago if I remember correctly.

closed 2025-02-21 by AlexandreRouma