AlexandreRouma/SDRPlusPlus #1562: NO SOUND SDR++ MAC M1

fgarciav1, 2025-01-02:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: MAC M1
- RAM:32Gb
- GPU: 
- SDR: (Remote or local? If remote, what protocol?)  RTL-SDR USB

**Software** SONOMA 14.6.1 
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15) 
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software). 1.2.1

**Bug Description**
A clear description of the bug.
The top speaker is set to "non-operational" and the slider bar cannot be activated.
The speaker bar at the bottom can slide.
THERE IS NO SOUND.
AM, USB, etc, etc reception modes are locked and cannot be selected.
**Steps To Reproduce**
1. ...
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.
With another program the rtl-sdr works well, with sound.
I have tried everything ChatGpt told me, uninstall, add repositories, various checks.
Everything turns out correctly, but the sound is NOT activated in any way, although the waterfall and the visual part work perfectly.
I thank you very much for your help.
Thanks and greetings.
[image]

fgarciav1, 2025-01-03:
Sorry in advance if I don't follow your instructions correctly, my computer skills are not very good.
If you need anything else, please let me know in detail if possible.
I copied the sdr++ version by clicking on the top icon on the right side, it is:

I tried to put the end of your instructions and this is what comes out.

Last login: Fri Jan  3 18:17:37 on ttys000

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
MacBook-Pro-de-FRANCISCO:~ FRANCISCO$ /Applications/SDR++.app/Contents/MacOS/sdrpp -c
[03/01/2025 18:23:58.000] [INFO] SDR++ v1.2.1
Unknown argument
-a --addr		Server mode address
   --autostart		Automatically start the SDR after loading
-h --help		Show help
-p --port		Server mode port
-r --root		Root directory, where all config files are stored
-s --server		Run in server mode
MacBook-Pro-de-FRANCISCO:~ FRANCISCO$ 

I have also made a screenshot in which you can see that it receives well, the spectrum is correct, but the upper speaker is blocked and the modes too.

[image]

Thanks por all

dh1tw, 2025-01-03:
looks like you are muted? 

Have you selected the correct sink / speakers? Check in the left menu "Sinks"

AlexandreRouma, 2025-01-03:
the checkbox on the radio was disabled.

closed 2025-01-03 by AlexandreRouma