AlexandreRouma/SDRPlusPlus #1563: the checkbox on the radio was disabled.

fgarciav1, 2025-01-04:
the checkbox on the radio was disabled.

_Originally posted by @AlexandreRouma in https://github.com/AlexandreRouma/SDRPlusPlus/issues/1562#issuecomment-2569824106_

fgarciav1, 2025-01-04:
Exactly, the checkbox is disabled but I can't enable it, that's the problem.
I see that my topic is closed, but I haven't understood well what I should do.
Thanks.

KubaPro010, 2025-01-04:
what happens when you try and why you allegedly can't?

fgarciav1, 2025-01-04:
As you can see in the photo, the top speaker is disabled and does not respond to anything, that is, when you click on it it does nothing, it remains with the disabled bar.
The bottom one is fine and the slider moves, but you can't hear it either.
The program works fine, it tunes the frequencies, the spectrum looks fine, etc., but there is no sound.
You can't change the reception modes either, even if you click on them.
The device works correctly with another program, that is, it must be some configuration problem, but I don't know any more.
The speakers are also selected correctly,
Thanks for the help

[image]
[image]

KubaPro010, 2025-01-04:
No, what happens when you try to click that checkbox right to `Radio` above `LSB`

fgarciav1, 2025-01-04:
I'm sorry, I'm so sorry, it was as easy as that, I couldn't see that checkbox and now, with your instructions, I've seen it. 
It was too shaded (the checkbox) and I couldn't see it.
It works perfectly.
It's silly but I've been doing this for two days.
I guess it's a beginner's mistake.
Thanks again.

KubaPro010, 2025-01-04:
close this issue then

closed 2025-01-04 by AlexandreRouma