AlexandreRouma/SDRPlusPlus #1565: RPiOS Bullseye: error while loading shared libraries: libvolk.so.2.4

RyanHir, 2025-01-07:
**Hardware**
- CPU: Broadcom BCM2837B0
- RAM: 1 GB
- GPU: N/A
- SDR: Local RTL-SDR Blog v4

**Software**
- Operating System: Raspberry Pi OS Bullseye armhf
- SDR++: v1.2.1 20:22:47 Jan 5, 2024

**Bug Description**
Loading SDR++ on the Pi results in failure due to dependency mismatch

**Steps To Reproduce**
1. Download `sdrpp_raspios_bullseye_armhf.deb` from nightly download
2. Install to Raspberry Pi via `sudo apt install ./sdrpp_raspios_bullseye_armhf.deb`
3. Run `sdrpp` or `sdrpp --server`

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**
```
pi@rtl:~ $ sdrpp -c
sdrpp: error while loading shared libraries: libvolk.so.2.4: cannot open shared object file: No such file or directory
```

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

KubaPro010, 2025-01-07:
try `sudo ldconfig`

RyanHir, 2025-01-07:
```
pi@rtl:~ $ sudo ldconfig 
pi@rtl:~ $ ldd /usr/bin/sdrpp 
	linux-vdso.so.1 (0x7eea8000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76ee0000)
	libsdrpp_core.so => /lib/libsdrpp_core.so (0x768d0000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76757000)
	libglfw.so.3 => /lib/arm-linux-gnueabihf/libglfw.so.3 (0x766e0000)
	libOpenGL.so.0 => /lib/arm-linux-gnueabihf/libOpenGL.so.0 (0x76680000)
	libfftw3f.so.3 => /lib/arm-linux-gnueabihf/libfftw3f.so.3 (0x76540000)
	libvolk.so.2.4 => not found
	libzstd.so.1 => /lib/arm-linux-gnueabihf/libzstd.so.1 (0x764a1000)
	libstdc++.so.6 => /lib/arm-linux-gnueabihf/libstdc++.so.6 (0x762d0000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76289000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76250000)
	/lib/ld-linux-armhf.so.3 (0x76f04000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76efd000)
	libX11.so.6 => /lib/arm-linux-gnueabihf/libX11.so.6 (0x760ec000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76efa000)
	libGLdispatch.so.0 => /lib/arm-linux-gnueabihf/libGLdispatch.so.0 (0x75fb0000)
	libxcb.so.1 => /lib/arm-linux-gnueabihf/libxcb.so.1 (0x75f70000)
	libXau.so.6 => /lib/arm-linux-gnueabihf/libXau.so.6 (0x75f50000)
	libXdmcp.so.6 => /lib/arm-linux-gnueabihf/libXdmcp.so.6 (0x75f30000)
	libbsd.so.0 => /lib/arm-linux-gnueabihf/libbsd.so.0 (0x75f00000)
	libmd.so.0 => /lib/arm-linux-gnueabihf/libmd.so.0 (0x75ee0000)
```

RyanHir, 2025-01-07:
Oh im sorry this is a problem on my end. I confused Bullseye and Bookworm.

closed 2025-01-07 by RyanHir