AlexandreRouma/SDRPlusPlus #1566: No sdrpp modules found when running linux arm64 build from source

telecter, 2025-01-07:
**Hardware**
I'm using Asahi Linux (arm64) on an M1 Mac.
- CPU: Apple M1
- RAM: 8gb
- GPU: Internal
- SDR: Local RTL-SDR Blog v4

**Software**
- Operating System: Fedora 41
- SDR++: 1.2.1 built at 10:49:25 on Jan 7 2025

**Bug Description**
When just running SDR++ using the command line, it fails because it is missing libsdrpp.
`sdrpp: error while loading shared libraries: libsdrpp_core.so: cannot open shared object file: No such file or directory`

Then, adding the plugins directory and `/usr/lib` to `LD_LIBRARY_PATH` allows SDR++ to start, but it still fails to find its modules.
`LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib:/usr/lib/sdrpp/plugins sdrpp`
In the logs:
```[07/01/2025 10:52:26.000] [INFO] Initializing Airspy Source (airspy_source)
[07/01/2025 10:52:26.000] [ERROR] Module 'airspy_source' doesn't exist
[07/01/2025 10:52:26.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[07/01/2025 10:52:26.000] [ERROR] Module 'airspyhf_source' doesn't exist
[... for every module]
```

**Steps To Reproduce**
I realise this may be specific to my M1 linux platform, and so may be difficult to reproduce.
1. Download source and build SDR++
4. Install SDR++ using `sudo make install`
5. Run `sdrpp` with and without adding `/usr/lib` and the plugins dir to `LD_LIBRARY_PATH` and observe logs

AlexandreRouma, 2025-01-07:
you didn't at all follow the MacOS instructions... You're not supposed to make install, you're supposed to create a bundle.
https://github.com/AlexandreRouma/SDRPlusPlus?tab=readme-ov-file#building-on-macos

telecter, 2025-01-07:
No, I am running Fedora Linux on an aarch64 platform. The hardware I am on is a MacBook.

AlexandreRouma, 2025-01-07:
It's an issue with builds on fedora then. They use nonsensical directories for everything while SDR++ was hardcoded for normal POSIX install paths (ie. no lib64 bullshit). It may get fixed later when I get around to finish the `better_install` branch but no ETA.

treybgreen, 2025-06-21:
The problem is that the dynamic linker isn't aware of the newly installed libraries after `sudo make install`. Run `sudo ldconfig` and then the app starts.

braydenplee, 2025-08-02:
@treybgreen 's solution works. Confirmed working on OpenSUSE Tumbleweed after building from source. @AlexandreRouma You can probably close this thread as solved with the above comment as the solution.