AlexandreRouma/SDRPlusPlus #1568: MacOS ARM builds are missing m17_decoder

jancona, 2025-01-10:
**Hardware**
- CPU: Apple M3
- RAM: 24 GB
- GPU: 
- SDR: RTL-SDR

**Software**
- Operating System: MacOS Sequoia 15.2
- SDR++: SDR++ v1.2.1 20:22:47, Jan 5 2025

**Bug Description**
The M17 Decoder is not packaged with the release. See Discord [discussion](https://discord.com/channels/752669522883903601/752672743690076284/1326995597500354744).

**Steps To Reproduce**
1. Run SDR++
2. Attempt to select m17_decoder in the Module Manager drop-down list, but it's not there.

**Additional info**
The m17_decoder was removed from the MacOS ARM build because of build failures like this:
```
In file included from /Users/runner/work/SDRPlusPlus/SDRPlusPlus/decoder_modules/m17_decoder/src/m17dsp.h:8:
/opt/homebrew/Cellar/codec2/1.2.0/include/codec2/codec2.h:31:10: fatal error: 'codec2/version.h' file not found
#include <codec2/version.h>
         ^~~~~~~~~~~~~~~~~~
1 warning and 1 error generated.
```
Because the MacOS Intel build is unaffected, this may be related to the ARM version of Homebrew being installed in `/opt/homebrew` instead of `/usr/local` on Intel Macs. I was able to get a clean build by defining `CPATH` in the build environment. See https://github.com/jancona/SDRPlusPlus/commit/e85f3fbfe4d9b02f3ab3e76f020698832aba7e82. There's probably a cleaner solution involving CMake and/or Homebrew, but finding it is beyond my skills.

om0wt, 2025-05-15:
My workaround:

```
➜  SDRPlusPlus git:(master) ✗ cd /opt/homebrew/Cellar/codec2/1.2.0/include/codec2
➜  codec2 git:(stable) ls -l
total 152
drwxr-xr-x@ 2 pavolcalfa  admin     64 15 máj 12:02 cdcodec2
drwxr-xr-x@ 3 pavolcalfa  admin     96 15 máj 11:50 codec2
-rw-r--r--@ 1 pavolcalfa  admin   4104 20 júl  2023 codec2.h
-rw-r--r--@ 1 pavolcalfa  admin   2970 20 júl  2023 codec2_cohpsk.h
-rw-r--r--@ 1 pavolcalfa  admin   4989 20 júl  2023 codec2_fdmdv.h
-rw-r--r--@ 1 pavolcalfa  admin   1679 20 júl  2023 codec2_fifo.h
-rw-r--r--@ 1 pavolcalfa  admin   1642 20 júl  2023 codec2_fm.h
-rw-r--r--@ 1 pavolcalfa  admin   3471 20 júl  2023 codec2_math.h
-rw-r--r--@ 1 pavolcalfa  admin   3197 20 júl  2023 codec2_ofdm.h
-rw-r--r--@ 1 pavolcalfa  admin   1034 20 júl  2023 comp.h
-rw-r--r--@ 1 pavolcalfa  admin  13984 20 júl  2023 freedv_api.h
-rw-r--r--@ 1 pavolcalfa  admin   7808 20 júl  2023 fsk.h
-rw-r--r--@ 1 pavolcalfa  admin   2975 20 júl  2023 modem_stats.h
-rw-r--r--@ 1 pavolcalfa  admin   2343 20 júl  2023 reliable_text.h
-rw-r--r--@ 1 pavolcalfa  admin   1215 20 júl  2023 version.h
➜  codec2 git:(stable) mkdir codec2
➜  codec2 git:(stable) cd codec2
➜  codec2 git:(stable) ln -sf ../version.h .
➜  codec2 git:(stable) ls -la
total 0
drwxr-xr-x@  3 pavolcalfa  admin   96 15 máj 12:02 .
drwxr-xr-x@ 16 pavolcalfa  admin  512 15 máj 12:02 ..
lrwxr-xr-x@  1 pavolcalfa  admin   12 15 máj 12:02 version.h -> ../version.h
```

Than it compiles with M17 decoder

socheatsok78, 2025-09-05:
We can can set these environment variables for the compiler. This sould build `m17_decoder ` module.

```sh
export CPLUS_INCLUDE_PATH=/opt/homebrew/include:$CPLUS_INCLUDE_PATH
export LIBRARY_PATH=/opt/homebrew/lib:$LIBRARY_PATH
```

```diff
diff --git a/.github/workflows/build_all.yml b/.github/workflows/build_all.yml
index aab90f35..1d12ffef 100644
--- a/.github/workflows/build_all.yml
+++ b/.github/workflows/build_all.yml
@@ -204,11 +204,14 @@ jobs:
 
         - name: Prepare CMake
           working-directory: ${{runner.workspace}}/build
-          run: cmake -DCMAKE_OSX_DEPLOYMENT_TARGET=10.15 $GITHUB_WORKSPACE -DOPT_BUILD_PLUTOSDR_SOURCE=ON -DOPT_BUILD_BLADERF_SOURCE=ON -DOPT_BUILD_SDRPLAY_SOURCE=ON -DOPT_BUILD_LIMESDR_SOURCE=ON -DOPT_BUILD_AUDIO_SINK=OFF -DOPT_BUILD_PORTAUDIO_SINK=ON -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON -DOPT_BUILD_M17_DECODER=OFF -DOPT_BUILD_PERSEUS_SOURCE=OFF -DOPT_BUILD_AUDIO_SOURCE=OFF -DOPT_BUILD_RFNM_SOURCE=ON -DOPT_BUILD_FOBOSSDR_SOURCE=ON -DOPT_BUILD_HYDRASDR_SOURCE=ON -DUSE_BUNDLE_DEFAULTS=ON -DCMAKE_BUILD_TYPE=Release
+          run: cmake -DCMAKE_OSX_DEPLOYMENT_TARGET=10.15 $GITHUB_WORKSPACE -DOPT_BUILD_PLUTOSDR_SOURCE=ON -DOPT_BUILD_BLADERF_SOURCE=ON -DOPT_BUILD_SDRPLAY_SOURCE=ON -DOPT_BUILD_LIMESDR_SOURCE=ON -DOPT_BUILD_AUDIO_SINK=OFF -DOPT_BUILD_PORTAUDIO_SINK=ON -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON -DOPT_BUILD_M17_DECODER=ON -DOPT_BUILD_PERSEUS_SOURCE=OFF -DOPT_BUILD_AUDIO_SOURCE=OFF -DOPT_BUILD_RFNM_SOURCE=ON -DOPT_BUILD_FOBOSSDR_SOURCE=ON -DOPT_BUILD_HYDRASDR_SOURCE=ON -DUSE_BUNDLE_DEFAULTS=ON -DCMAKE_BUILD_TYPE=Release
 
         - name: Build
           working-directory: ${{runner.workspace}}/build
-          run: make VERBOSE=1 -j3
+          run: |
+              export CPLUS_INCLUDE_PATH=/opt/homebrew/include:$CPLUS_INCLUDE_PATH
+              export LIBRARY_PATH=/opt/homebrew/lib:$LIBRARY_PATH
+              make VERBOSE=1 -j3
 
         - name: Create Archive
           working-directory: ${{runner.workspace}}
```

AlexandreRouma, 2025-09-05:
It was disabled because of codec2 build issues on MacOS. You should not have to set environment variables.

socheatsok78, 2025-09-05:
Apology if i misunderstood, was it the error OP mentioned above about `'codec2/version.h' file not found`?

jancona, 2026-11-06:
The Homebrew side is fixed: codec2’s pkg-config file now exports both include and include/codec2 ([Homebrew/homebrew-core#323804](https://github.com/Homebrew/homebrew-core/issues/323804), a backport of [drowe67/codec2#49](https://github.com/drowe67/codec2/issues/49)), and rebuilt bottles are available. With current Homebrew codec2 on Apple Silicon, building with -DOPT_BUILD_M17_DECODER=ON works without setting CPATH or any other environment variables, so M17 could be re-enabled in the macOS ARM CI build.