AlexandreRouma/SDRPlusPlus #1576: SDR++ is not available in the application menu on KDE

SrS2225a, 2025-02-02:
**Hardware**
- CPU: 12 × Intel® Core™ i7-8700 CPU @ 3.20GHz
- RAM: 31.2 GiB of RAM
- GPU: NVIDIA GeForce GTX 1050 Ti/PCIe/SSE2
- SDR: Local

**Software**
- Operating System: KDE Plasma v6.2.5 with Kernel 6.12.10-arch1-1 (64-bit) 
- SDR++: SDR++ v1.2.1 (Built at 20:34:00, Feb 2 2025)

**Bug Description**
I am not sure about other desktops, but on KDE at least, when you open the "Application Menu" (the main way to open applications and find them), there is no application for SDR++ listed. Instead, I have to launch it manually from the terminal or locate the binary manually. This issue is caused by an incomplete/missing Categories entry in the .desktop file.

**Steps To Reproduce**
1. Install SDR++ on KDE Plasma.
2. Open the application menu and search for "SDR++".
3. Notice that SDR++ is not listed in the search results

**Expected Behavior**
SDR++ should be available in the application menu, properly categorized, so users can launch it without needing the terminal.

**Solution / Workaround**
Manually updating or creating the .desktop file in `/usr/share/applications/sdrpp.desktop` with the following content resolves the issue:

```ini
[Desktop Entry]
Encoding=UTF-8
Version=1.0
Type=Application
Terminal=false
Exec=/usr/bin/sdrpp
Name=SDR++
GenericName=Software Defined Radio
Comment=Cross-Platform SDR Software
Icon=/usr/share/sdrpp/icons/sdrpp.png
Categories=AudioVideo;Audio;HamRadio;
Keywords=SDR;Radio;HAM;
```

After making this change and running update-desktop-database, SDR++ appears in the application menu as expected. It is also worth mentioning, it would also be acceptable to add `GenericName`, `Comment`, and `Keywords` as the desktop file is missing that key metadata; it is not required but helps with providing more information so it is easier to find in an application menu

**Screenshots**
Before change:
[image]

After change:
[image]

**Additional info**
1. Running `which sdrpp` confirms the binary is installed.
2. Checking `/usr/share/applications/` reveals an incorrect `sdrpp.desktop` file.
3. Other applications installed on the system appear in the application menu as expected.
4. **Since the project does not accept code pull requests, would it be acceptable to submit a pull request for this fix, or should it be left as an issue for maintainers to resolve?**

AlexandreRouma, 2025-02-02:
I don't know how you installed it but I'm on KDE plasma and it's always shown up fine...
nothing has ever changed with how the the .desktop is installed in over 4 years and nobody has reported any problem...

Definitely a problem on your side.

closed 2025-02-02 by AlexandreRouma

SrS2225a, 2025-02-03:
🤔 I see, how odd indeed, are you sure? I used the AUR package sdrpp-git. I even tried building it from source with the instructions listed in the read me with the same results. Are you using the latest KDE version v6.2.5?  I think possibly the reason why no one ever reported on it is that you can still find it just fine if you use the search bar. Plus, not everyone uses KDE

Anyways, it could be an issue on my side, but I dont think so because every other application I have shows up correctly just fine, and SDRPP is the only exception where it doesn't, and I do have all the latest packages for Arch, KDE and SDRPP. Regardless, I will see if I can investigate this more.

Even then if it is an issue on my end, I would still highly recommend that you add those `GenericName`, `Comment`, and `Keywords` fields in the desktop entry since it is missing that key metadata, which would be greatly appreciated. Thank you for taking the time to take a look at this!

SrS2225a, 2025-02-21:
So, are you actually going to implement my suggestions? Or pretend like my issue does not exist? Because remember: I had to fix it with some minor adjustments the the desktop file. "Works on my machine" is not a valid argument because it disregards the complexities of different environments, dependencies, and configurations that can affect software behavior

AlexandreRouma, 2025-02-21:
It's not a case of "works on my machine", it's a case of "works on everyone else's machine but yours".
I'm not changing the desktop file that already works for everyone and have to retest it just for one person.
If other people had that issue, someone else would have come forward by now in the 5 years this exact .desktop file has been used...

SrS2225a, 2025-02-21:
I get that. Sorry about that, your last comment came off as a bit rude to me, so sorry if I came off a little defensive. 

I understand that if this issue were widespread, others would have reported it by now. However, I still think it's worth considering minor improvements to the `.desktop` file, which improve discoverability and usability.

Even though the issue might be on my end, having these fields would still benefit all users by making the application easier to find and more consistent with other desktop entries. Would you be open to a pull request specifically for those metadata additions? It wouldn’t change any of the existing behavior but would enhance the `.desktop` file for those who rely on search and categorization.

Thanks for taking the time to respond and for maintaining SDR++!