AlexandreRouma/SDRPlusPlus #1580: Tezuka Firmware for the Pluto source

Notupus, 2025-02-05:
Make so sdrpp handles the tezuka firmware compatible with most plutoish devices in such a way to fully utilize the capabilities of the hardware.
Basically https://github.com/AlexandreRouma/SDRPlusPlus/pull/1512 
But in issues.

Notupus, 2025-02-05:
Okay first of all, for tezuka and any firmware you can use rx2 by setting the following reg write
RX1
iio_device_reg_write(_this->phy, 0x00000003, 0x40);
RX2
iio_device_reg_write(_this->phy, 0x00000003, 0x80);
In order for those reg writes to work you need to also set the 1rx 1tx mode based on rx number
iio_device_debug_attr_write_longlong(_this->phy,"adi,1rx-1tx-mode-use-rx-num",_this->rfId+1);
rfid is 0 for rx1 and rfid is 1 for rx2
This seems redundant but its the only way it has worked.

For the 8 bit mode in the tezuka firmware you need to only disable the rx0_q stream and my understanding is that makes tezuka do a iq stream over the 8 bits the i channel.
The rest is using the right volk function for 8bits to 32bits and it resets the buffer on the pluto device if 
iio_device_reg_read(_this->dev, 0x80000088, &val); bit 2 is 1 and if so it clears the adc underflow bits, basically this is just another thing that shows stuff in the log and I dont think its critical for sdrpp but a nice to have.

AlexandreRouma, 2025-02-05:
Ok so this is a complete and utter hack...
Writing registers directly like that is a recipe for disaster when libad9361/libiio expects to be in control of them.

This is gonna be a no from me until it can be done properly through IIO and the firmware takes care of setting these registers/attributes. I'm not risking the stability of the PlutoSDR source module for a tiny number of users on some half baked third party firmware.