AlexandreRouma/SDRPlusPlus #1581: Added plugin not shown in ModuleManager

StefanKDS, 2025-02-05:
Hi,

I created a plugin for SDR++. 
I compiled everything and everything is working.

Now I copied the plugin .so file to another computer which SDR++ has installed ( into the plugin folder ),
but my plugin is not shown in the ModuleManager.

I working on Linux.

As far as I understood it should be enough to copy the module / plugin .so file, isn't it ?

I want to release these an other plugins for SDR++ but I need to know how a user can add these to his SDR++.

Regards
Stefan

AlexandreRouma, 2025-02-05:
Copying them should work provided the module does not rely on dependencies not installed on the second machine.

The logs are printed directly to terminal when you run `sdrpp`.

AlexandreRouma, 2025-02-19:
closed for inactivity

closed 2025-02-19 by AlexandreRouma