AlexandreRouma/SDRPlusPlus #1592: Failed to execute child process

KC3TDR, 2025-03-06:
System is Raspberry Pi 4B running Raspberry Pi OS( 64bit) Debian Bookworm
Downoaded and installed Nightly Build V1.2.1 of sdrpp_debian_bookworm_aarch64.deb.
Consistently get error: Failed to execute child process "/usr/bin/sdrpp" (no such file or directory).
Any help or suggestion would be greatly appreciated.
73
Frank KC3TDR

AlexandreRouma, 2025-03-06:
Show the output of `file /usr/bin/sdrpp`.

KC3TDR, 2025-03-06:
Please explain what you mean by "show the output of file
/usr/bin/sdrpp". I'm not a linux power user. File is not ASCII.

Frank

AlexandreRouma, 2025-03-06:
Run the command `file /usr/bin/sdrpp` in a terminal and show what it says.

KC3TDR, 2025-03-06:
bash: /usr/bin/sdrpp: cannot execute: required file not found

AlexandreRouma, 2025-03-06:
You didn't run the correct command. Type word for word `file /usr/bin/sdrpp` in a terminal and press enter. The `file ` part at the beginning is important.

KC3TDR, 2025-03-06:
Same response. entered /usr/bin/sdrpp from home directory

AlexandreRouma, 2025-03-06:
You should type and see something like this

[image]

KC3TDR, 2025-03-06:
> You should type and see something like this
>
> image.png (view on web)

file /usr/bin/sdrpp

/usr/bin/sdrpp: ELF 64-bit LSB pie executable. ARM aarch64. version 1
(SYSV). dynamically linked. interpreter /lib/id-linux-aarch64.so.1.
BuildID[sha1}=bf7b7d5711faf7752021793ed27fcaf939fc3f1c. for GNU/Linux
3.7.0. not stripped

AlexandreRouma, 2025-03-06:
Ok and now run this and show the output

[image]

KC3TDR, 2025-03-06:
> Ok and now run this and show the output
>
> image.png (view on web)

Yes that's correct

AlexandreRouma, 2025-03-06:
ok so your raspian install is 32bit, not 64bit like you thought.

KC3TDR, 2025-03-06:
Don't understand that I used Raspberry Pi Imager, chose pi 4 and top os
listed as 64bit bookworm recommended. I'll try reimaging, unless you have
any other followup?

Thanks
Frank

AlexandreRouma, 2025-03-06:
`uname -m` tells you the hardware architecture, not the OS architecture. uname will always show `aarch64` when run on a pi4 no matter what version of raspios you put on it.

KC3TDR, 2025-03-06:
Ok, thanks for the clarification.

KC3TDR, 2025-03-06:
Ok. I got it to work. I did a reimage and reinstall, took my time. Must
have been a typo somewhere along the way.

Thanks for all your good work and help.
Frank

closed 2025-03-06 by AlexandreRouma