AlexandreRouma/SDRPlusPlus #1594: Ignore duplicate IP based sources that have same serial on Pluto Source

alphafox02, 2025-03-11:
As discussed in discord, sorry for lack of details, but in 24.04 libiio/pluto release now takes on an ipv4 & ipv6. SDRPlusPlus crashes when both ips are present as they both come up as the same serial and type of source. 

I tested a minimal change to the source block that skips a duplicate ip based entry. SDRPlusPlus starts fine and presents the user one ip based entry and one usb entry per Pluto.

https://github.com/alphafox02/SDRPlusPlus/commit/99e20e9d055598ba687d10c2223056653804c0cd

povnz, 2025-03-22:
Thank you for highlighting this, i was trying to figure out why it kept crashing, I'm going to attempt to disable IPv6 as a work around

alphafox02, 2025-03-22:
That works too. When the interface for Pluto comes up, go into network manager and disable ipv6. Then open up SDR++.

povnz, 2025-03-23:
Confirmed it works great thanks Alphafox02

nmaas87, 2025-04-28:
@alphafox02 is there a reason as to why you did not integrate it as a PR to @AlexandreRouma repo if it does resolves the issue?
Thanks a lot for the fix 👍🏻

alphafox02, 2025-04-28:
I thought I did, I’ll go back and look. I did mention it to him, the developer, but maybe I forgot to make a PR.

AlexandreRouma, 2025-04-30:
Should be fixed, let me know if not.

closed 2025-04-30 by AlexandreRouma