AlexandreRouma/SDRPlusPlus #1605: Radio plugin: (option to) remove bandwidth limits of VFO

ckahlo, 2025-03-21:
Currently SDR++ follows a concept also seen e.g. in SDR-Console to limit the maximum bandwidth
to quite small values depending on the modulation selected:
AM = 15000Hz, LSB/USB/DSB = 12000Hz, NFM = 50000Hz, WFM = 250000Hz

However, wider signals may exist. Especially if the signal contains multiple transponders and has to
be post-processed by another decoder it is quite typical to use USB with ~96, ~192, ~384 kHz and
a network sink or virtual (or digital physical) audio connection between SDR and decoder without
streaming the whole (much larger) baseband signal.

SDR# has some defaults depending on the modulation, but does not limit the user to select much
higher bandwidths. As many people here I would like to get rid of SDR# and convince others as well.

Improvement: remove or at least heavily increase the bandwidth selection limits.

AlexandreRouma, 2025-03-21:
It is a clear that you have a fundamental misunderstanding of how SDRs work, otherwise you would know why that "limit" exists. The demodulators run at a fixed samplerate which determins their maximum bandwidth. That's all. Running them faster would waste CPU for everyone, and adjusting the samplerate dynamically is a bad idea.

> USB with ~96, ~192, ~384 kHz

Unless you're a bat and can hear ultrasound, no, there is no such thing as 96KHz SSB. You're trying to misuse a SSB demodulator to receive things that  aren't SSB. Use the raw demodulator or the IQ exporter module instead.

closed 2025-03-21 by AlexandreRouma

ckahlo, 2025-03-21:
The sample rates are adjustable already, so everyone could you just use the defaults or the (lower) samplerate needed.
So, no one is going to waste any CPU time. And I didn't say "dynamically", I was referring to the radio settings.
As stated before I was talking about transponders and secondary decoders - NOT listening to audio(!). The IQ exporter is a valid point indeed, but the 2 secondary decoders I cannot change - because I don't have code - don't support network sources.
I am just trying to find a way to solve a specific scenario following the decoder instructions. Using the RAW demodulator didn't work out as expected. I got no signal lock, but I can try again.
Every transponder is a few kHz wide, every 10kHz, and with SDR# I used to stream 7 of them in an audio loopback towards the decoders. As this was working fine before I thougt this would be easy with SDR++.

Well, I joined Patreon for this project, but the aggressive ad hominem tone in reference to this question/request is just mind-boggling. Then SDR# still is the choice for this use case.
Thank you very much for your answer.

AlexandreRouma, 2025-03-21:
> The sample rates are adjustable already, so everyone could you just use the defaults or the (lower) samplerate needed.

They are not, never have been and never will be. The demodulators do not run at the same samplerate as the SDR itself. Data is channelized and resampled before it ever gets to the radio, and that samplerate limits the bandwidth.

ckahlo, 2025-03-21:
OK, so we're getting closer.
The architectural design decision about the re-sampling design is specific to _your_ SDR and inherently limits the decoder bandwidth - in contrast to other approaches. Optimizing CPU usage is the reason. Thanks for explanation.

So, this will never change in SDR++ as you say and as long as I don't write an IQ Exporter to audio loopback component we should stay with SDR#. No need to blame people as clueless.

As you're studying aerospace engineering then have a look at INMARSAT. That's the usecase. 
Greetings from team MAPHEUS, sounding rockets at ESRANGE. Just Google.