AlexandreRouma/SDRPlusPlus #1617: Segmentation fault in SDR++ server on Pi 2B when changing antenna setting

fat-tony, 2025-04-30:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: 
- RAM:
- GPU: 
- SDR: (Remote or local? If remote, what protocol?)

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

**Bug Description**
A clear description of the bug.

**Steps To Reproduce**
1. ...
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

fat-tony, 2025-04-30:
Running SDR++ 1.2.1 in server mode built from source yesterday. Running on headless RPi2 B 32 bit bookworm. SDRPlay 3.15 API into an RSP2.
Client is SDR++ 1.2.1 nightly built April 23rd running on x64 linux mint.

Runs ok with antenna port A or B selected,

When I select port High-Z the server crashes with segmentation fault.

[30/04/2025 11:30:43.000] [INFO] Running post-init for RTL-SDR Source
[30/04/2025 11:30:43.000] [INFO] Running post-init for RTL-TCP Source
[30/04/2025 11:30:43.000] [INFO] Running post-init for SDR++ Server Source
[30/04/2025 11:30:43.000] [INFO] Running post-init for SDRplay Source
[30/04/2025 11:30:43.000] [INFO] Running post-init for Spectran HTTP Source
[30/04/2025 11:30:43.000] [INFO] Running post-init for SpyServer Source
[30/04/2025 11:30:43.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Menu Select!
[30/04/2025 11:30:43.000] [INFO] Ready, listening on 0.0.0.0:5259
[30/04/2025 11:31:38.000] [INFO] Connection from TODO:TODO
[30/04/2025 11:32:08.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Menu Deselect!
[30/04/2025 11:32:08.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Menu Select!
Segmentation fault

AlexandreRouma, 2025-04-30:
Your raspberry pi is too old and is running out of ram.

closed 2025-04-30 by AlexandreRouma

fat-tony, 2025-05-01:
Hehe - I know how it feels to be too old and running out of RAM! 
I though that maybe I was testing the capability of the RPi too much but I'm running it headless and SDR++ is running only in server mode and I access from an x64 version of SDR++ on my PC. It runs for hours serving high-bandwidth FM stereo without issue or any other mode for that matter until I select Hi-Z antenna , so I thought that maybe it was just a bug in the code because the Hi-Z antenna is selected through AM_Port call, which is different to the antenna A and B.
 _top_ shows over 700MB of memory free immediately before the seg fault is triggered on the server when I select Hi-Z on the client. I thought I might re-build again from source and remove any superfluous interfaces as I'm literally just using an RSP2. Just to eliminate any confusion you might have, I reported this on Discord as Joe_90. This "fat-tony" handle is one I used many years ago. I attach screenshots of before and after the seg fault showing that memory "shouldn't" be the issue. 
I appreciate your time and effort Alexandre!

[image]
[image]

KubaPro010, 2025-05-01:
Can you try my fork of [SDR++ for server](https://github.com/KubaPro010/SDRMinusMinus)? clone it, uncomment the `#define SDRPlay` of `source_modules/mirisdr_source/src` and compile like sdr++? maybe it will work better