AlexandreRouma/SDRPlusPlus #1623: Mute on TX on panadpter use

nex2000, 2025-05-04:
I'm trying to get to know SDRpp better, it's a great program, I managed to use the cat using FLrig + CatRelay, I use SDRpp as a panoramic receiver, so combined with a radio that transmits, how do I have the mute on Tx, when I go to TX I see a big signal on the spectrum and I hear my voice. How can I silence SDRpp during TX.

soakland5, 2025-05-15:
I am a ham operator and second that request. 

Nex2000, can you and I connect? I am transitioning from SDR Console on Windows to SDR++ on Mac, and would like to find out how you managed to get it all to work :) 73 de Scott N1IA

nex2000, 2025-05-16:
Hi Scott , my name is Paul my callsign is IT9CZM you can find my info on qrz.com 

unfortunately I realize that there are still no programs on Linux better than those that work on windows, there are many aspects that are still missing on SDRpp, manual notch, advanced NR, and others but there are good bases to continue development.

If you want information on SDR programs you can contact me I have tried everything but I have not yet found something complete that can have what is needed to be used as a transceiver.