AlexandreRouma/SDRPlusPlus #1624: CMake option to disable system wide iio and ad9361 framework

alexey-lysiuk, 2025-05-05:
Recent versions of libiio and libad9361-iio have options to disable macOS framework, so usual POSIX library installation is used instead.

Is it possible to add similar CMale option to SDR++? At the moment, PlutoSDR target tries to link with
system wide installation iio.framework and ad9361.framework.

AlexandreRouma, 2025-05-05:
no, you're not supposed to do that on macos.

closed 2025-05-05 by AlexandreRouma

alexey-lysiuk, 2025-05-05:
> no, you're not supposed to do that on macos.

Why? I don't want to install any library or framework into the system directories. I can build all dependencies from sources, and install them to an arbitrary directory. The root of this directory structure will be used as prefix directory for all dependencies.

I'm not asking to make this behavior the default one. I'm asking for an ability to customize this.