AlexandreRouma/SDRPlusPlus #1625: Startup on windows sometimes very very slow (loading modules)

chrthr, 2025-05-05:
**Hardware**
- CPU: Ryzen 7 Pro 7840U (thinkpad T14s)
- RAM: 32 GB
- GPU: Radeon 780M
- SDR: sdrpp server source -> rtlsdr @ raspi

**Software**
- Operating System: Win 11 Pro x64
- SDR++: 1.2.1 nightly build 13.04.25

**Bug Description**
Sometimes the application takes very long to start (40 seconds!) . Each module takes multiple seconds to load.
Usually it starts very fast (< 1 second) but in very few cases it seems to hang. I was yet unable to find any pattern when this will happen. 

You can see the startup splash screen slowly loading the modules and windows will offer you to terminate the program.

Example log:

```
C:\sdr\sdrpp_windows_x64_2025>sdrpp.exe -c
[05/05/2025 15:05:06.000] [INFO] SDR++ v1.2.1
[05/05/2025 15:05:06.000] [INFO] Loading config
[05/05/2025 15:05:06.000] [INFO] Using OpenGL 3.0
[05/05/2025 15:05:06.000] [INFO] Loading icons
[05/05/2025 15:05:07.000] [INFO] Loading band plans
[05/05/2025 15:05:07.000] [INFO] Loading band plans color table
[05/05/2025 15:05:07.000] [INFO] Loading modules
[05/05/2025 15:05:07.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/audio_sink.dll
[05/05/2025 15:05:08.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/audio_source.dll
[05/05/2025 15:05:08.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/file_source.dll
[05/05/2025 15:05:11.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/frequency_manager.dll
[05/05/2025 15:05:11.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/iq_exporter.dll
[05/05/2025 15:05:21.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/m17_decoder.dll
[05/05/2025 15:05:23.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/meteor_demodulator.dll
[05/05/2025 15:05:24.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/network_sink.dll
[05/05/2025 15:05:25.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/network_source.dll
[05/05/2025 15:05:26.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/plutosdr_source.dll
[05/05/2025 15:05:28.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/radio.dll
[05/05/2025 15:05:39.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/recorder.dll
[05/05/2025 15:05:40.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/rigctl_client.dll
[05/05/2025 15:05:41.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/rigctl_server.dll
[05/05/2025 15:05:41.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/rtl_sdr_source.dll
[05/05/2025 15:05:42.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/rtl_tcp_source.dll
[05/05/2025 15:05:43.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/scanner.dll
[05/05/2025 15:05:43.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/sdrplay_source.dll
[05/05/2025 15:05:44.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/sdrpp_server_source.dll
[05/05/2025 15:05:46.000] [INFO] Loading C:/sdr/sdrpp_windows_x64_2025/modules/spyserver_source.dll
[05/05/2025 15:05:47.000] [INFO] Initializing Audio Sink (audio_sink)
[05/05/2025 15:05:47.000] [INFO] Initializing Audio Source (audio_source)
[05/05/2025 15:05:47.000] [INFO] Initializing File Source (file_source)
[05/05/2025 15:05:47.000] [INFO] Initializing Frequency Manager (frequency_manager)
[05/05/2025 15:05:47.000] [INFO] Initializing Network Sink (network_sink)
[05/05/2025 15:05:47.000] [INFO] Initializing Network Source (network_source)
[05/05/2025 15:05:47.000] [INFO] Initializing PlutoSDR Source (plutosdr_source)
[05/05/2025 15:05:47.000] [INFO] Initializing RTL-SDR Source (rtl_sdr_source)
[05/05/2025 15:05:47.000] [INFO] Initializing RTL-TCP Source (rtl_tcp_source)
[05/05/2025 15:05:47.000] [INFO] Initializing Radio (radio)
```

**Steps To Reproduce**
1. Start application
2. Measure startup time

**Additional info**
I already removed some modules to speed up the startup process. Still the remaining modules take a long time to load sometimes.

AlexandreRouma, 2025-05-05:
Not anything within SDR++'s control. It's something libusb sometimes does.

closed 2025-05-05 by AlexandreRouma

AlexandreRouma, 2025-05-05:
Nearly all SDR drivers/control library use libusb.

chrthr, 2025-05-05:
The bundled version of libusb in sdrpp windows build is :  1.0.23.11397 (which is more than 5 years old?)

Could it help to replace it with a more recent version? https://github.com/libusb/libusb/releases

Anyway I also wanted to say thanks for this excellent software!

AlexandreRouma, 2025-05-05:
It's a known-good version. Update at your own risk, I'm not changing the version that ships with SDR++.