AlexandreRouma/SDRPlusPlus #1626: Doesn't work on a virtual machine

Maxim717, 2025-05-06:
Doesn't work on a virtual machine, complains about OpenGL — is there a way to bypass this?

AlexandreRouma, 2025-05-06:
There is not, OpenGL is required.

closed 2025-05-06 by AlexandreRouma

johntdavis84, 2025-05-06:
> Maxim717
>  created an issue

What OS are you running on the virtual machine?  Depending on the OS, there should be a package available that includes OpenGL.

bk4nt, 2025-05-24:
One needs a wrapper to support this. This is what I'm willing to configure, sdrpp running elsewhere, preferably inside a VM. And why I ended creating #1639

https://github.com/AlexandreRouma/SDRPlusPlus/issues/1639

bk4nt, 2025-06-20:
CPU rendering is very poor. I've a Windows 11 installed on bare I7 6C/12T. On which I connect using RDP, which partially only supports GL. I can run there some (not all) GL demanding apps using an RDP session, one of those being SDR++. This stressing/loading a lot the I7, audio is crackling (drops, cutouts). Some of the 12T often being at 80% or more.

Inside a VM, this doesn't seem to work. I've now added a second GPU, attached/dedicated to a Windows WM via PCI pass through. An Nvidia GPU. Nvidia providing nvidiaopenglrdp.exe. That way, I'm getting GL and CUDA using RDP. So CPU/GPU loads are fine, SDR++ is fine, audio is good. Don't know if that could work with a cheap GT 730 or GT 710. The other GPU I have being that of the Ryzen CPU, kept for the KVM host.

Windows had RemoteFX vGPU. That feature was deactivated: https://support.microsoft.com/fr-fr/topic/kb4570006-mise-%C3%A0-jour-pour-d%C3%A9sactiver-et-supprimer-le-composant-vgpu-remotefx-dans-windows-bbdf1531-7188-2bf4-0de6-641de79f09d2#bkmk_alternatives

Maybe this could help: https://wiki.archlinux.org/title/QEMU/Guest_graphics_acceleration#

What I had recently tested being at end of page, 4, "[Venus](https://docs.mesa3d.org/drivers/venus.html) is a Virtio-GPU protocol for Vulkan command serialization. It is available since QEMU version 9.2.0 and Linux kernel version 6.13. See this [Github Gist](https://gist.github.com/peppergrayxyz/fdc9042760273d137dddd3e97034385f#file-qemu-vulkan-virtio-md) how to use it."