AlexandreRouma/SDRPlusPlus #1627: sdrpp crashes when any menu item is expanded or collapsed

wptracy, 2025-05-07:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: 
- RAM:
- GPU: 
- SDR: (Remote or local? If remote, what protocol?)

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

**Bug Description**
A clear description of the bug.

**Steps To Reproduce**
1. ...
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

AlexandreRouma, 2025-05-07:
Information not filled out.

closed 2025-05-07 by AlexandreRouma

wptracy, 2025-05-07:
sdrpp crashes when any menu item is expanded or collapsed

sub-menu items do collapse and expand without crashing sdrpp

I don't know when this started but I do know it used to work.

I've since been updating the os with sudo apt update, sudo apt full-upgrade.

I unplugged the SDRplay RSPdx and re-seated the RTL-SDR v4 and got the same result.

```
wptracy@Rpi5-3:~ $ sdrpp -v
[07/05/2025 13:55:32.000] [INFO] SDR++ v1.2.1

wptracy@Rpi5-3:~ $ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 1df7:3030 SDRplay RSPdx
Bus 003 Device 005: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
wptracy@Rpi5-3:~ $ 
```
I tried this on a different Raspberry Pi that has been disconnected for months.

It's running the bullseye OS and SDR++ v1.1.0.

SDR++ v1.1.0 works on the OS that has not be getting sudo apt updates.

I don't know how to determine which apt update broke it, or even if it was an apt update that broke it.

I have RTL-SDR v4 on two Rpi5's and one Rpi4.  It only works on the Rpi4 that has not been getting apt updates.

I know it used to work on the Rpi5's.

```
pi@64bit8GramRpiLogitecMediaServer:~ $ sdrpp -h
[07/05/2025 13:37:21.000] [INFO] SDR++ v1.1.0
-a --addr		Server mode address
   --autostart		Automatically start the SDR after loading
-h --help		Show help
-p --port		Server mode port
-r --root		Root directory, where all config files are stored
-s --server		Run in server mode
pi@64bit8GramRpiLogitecMediaServer:~ $ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
pi@64bit8GramRpiLogitecMediaServer:~ $ lsusb
Bus 002 Device 002: ID 0bda:9210 Realtek Semiconductor Corp. RTL9210 M.2 NVME Adapter
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@64bit8GramRpiLogitecMediaServer:~ $ 
```

[image]