AlexandreRouma/SDRPlusPlus #1631: Touchscreen support

jfedyczak, 2025-05-11:
**Feature description**

I've compiled SDR++ from source under most recent Ubuntu. I have a touchscreen (ThinkPad X12) and touch events are completely ignored in SDR++. Would it be possible to add even rudimentary support for touch events?

AlexandreRouma, 2025-05-11:
Most touch screens generate mouse inputs and SDR++ is known to work fine with those. GLFW (the GUI backend used by SDR++) has no touch-specific API.

closed 2025-05-11 by AlexandreRouma

jfedyczak, 2025-05-16:
I believe this is not the case under Wayland. Touch events are indeed translated to mouse events under X11 and even Windows, but Wayland on modern Linuxes does not seem to do this. I have the same issue under most recent Fedora. Here's similar issue reported in GLFW. But as I understand, you don't have any means to fix it unless GLFW starts to expose touch events.

AlexandreRouma, 2025-05-16:
In any case this is not something that "fix" at the application level. GLFW doesn't have a concept of touch events to begin with, so it's there that the conversion must be done if necessary.