AlexandreRouma/SDRPlusPlus #1634: High CPU useage when idle

PetiAPocok, 2025-05-18:
**Hardware**
- CPU: AMD Ryzen 7 3800X 8-Core Processor
- RAM: 32GB
- GPU: Radeon RX 6600 XT
- SDR: local RTL-SDR v4 / SDRplay

**Software**
- Operating System: Debian 12.11.0 (6.12.12-1~bpo12+1)
- SDR++: 1.2.1 + 2025.04.30

**Bug Description**
High CPU usage when doing nothing.

**Steps To Reproduce**
1. Start the software.
2. It's already doing it
3. (If you start the listening, it drops for a second.)

**Screenshots**
On the report PC:
[image]
On my weaker machine:
[image]

**Additional info**
None?

AlexandreRouma, 2025-05-18:
Your OpenGL driver is not working properly.

closed 2025-05-18 by AlexandreRouma

KubaPro010, 2025-05-18:
> Your OpenGL driver is not working properly.
Nope, this is cross-platform and occurs in windows too

PetiAPocok, 2025-05-23:
> How should I check? What should I do?
Use vainfo to see if openGL is really using any GPU.

bk4nt, 2025-05-24:
Try "glxinfo | grep -i opengl". The renderer shall be the graphic card. If not, the CPUs are getting busy...

[image]

PetiAPocok, 2025-06-01:
The vainfo shows more clearly that there are problems:

[image]

(The intel N processors' integrated GPUs doesn't really have a linux firmware.)

bk4nt, 2025-06-01:
When you see renderer llvmpipe, GLX is provided by CPU, not by GPU. So the CPU has a lot to do.

The intel N100 (of Q1 23) shall provide GL by hardware: https://www.intel.com/content/www/us/en/products/sku/231803/intel-processor-n100-6m-cache-up-to-3-40-ghz/specifications.html

6.12.12 is of nov last year. That should provide you the GPU drivers/firmwares.

See the output of "inxi -Fxxxrz", if graphics driver is i915. 

Run lspci | grep -i 'vga'. You will get a device ID like 05:00.0. Run and see the ouput of lspci -v -s 05:00.0.

See also in the output of dmesg, if you have any error related to graphics.

"Intel GPU hardware acceleration can be confirmed via intel_gpu_top found in intel-gpu-tools" package.

Something could be wrong with your current driver or settings for N100.

PetiAPocok, 2025-06-08:
So the issue was that Debian 12 (bookworm) has intel-media-va-driver (23.1.1+dfsg1-1) but intel-media-va-driver (25.2.3+dfsg1-1) has the support for the Intel N150's integrated GPU which is currently in Debian 13 (trixie) (testing). So after switching to testing, updating everything, the issue is solved.