AlexandreRouma/SDRPlusPlus #1635: M1 Mac no audio

soakland5, 2025-05-20:
I’m getting no audio from SDR++ on a Mac M1 machine. I’ve checked MIDI and my sound settings. I get a grayed-out speaker icon in SDR++ but am able to select different output devices, but nothing. These devices all work  for YouTube. I see signals on the display though.

I saw someone on here had a similar issue but unsure how it was resolved. I even tried the Intel build.

Anyone have any idea how to fix this ?

Scott N1IA

Kathytaff, 2025-07-12:
Same problem. It did work for a minute, then quit. Also cannot change other settings such as moving bands, bandwidth, squelch, etc.

Kathytaff, 2025-11-02:
scroll down on left hand side list of options. look for "radio". should have a blue check mark next to it. It's been a while, but this is what I remember doing. Alexandre Rouma suggested it somewhere.

closed 2026-10-22 by AlexandreRouma