AlexandreRouma/SDRPlusPlus #1641: SDR++ crash and recovery whilst using KVM/QEMU, x2go and VirtualGL

bk4nt, 2025-05-30:
Hello,

I understand such a setup can't be supported. VM/KVM with GLX is experimental. But x2go and VitualGL are widely used.

The main point being that after such a crash, SDR++ can't be restarted without a VM reboot. Because some files or IPCs seem still be locked by any process of the crashed SDR++ app. So far, I couldn't find what could remain locked by the crashed then killed SDR++ app.

Is there anything to check and kill before VM reboot?

Step to reproduce:
- run SDR++ in a VM, connecting to the VM using x2go and VirtualGL for GLX support
- use SDR++, which runs so very fine,
- resize SDR++, the GUI hangs (but the decoding and sound still remains fine),
- kill that crashed SDR++ app, launch a new one,
- the new one hangs at startup, at "Loading config"...

With another env (x2go and VirtualGL on a physical host) , I was getting the same or similar after such a crash, SDR++ then hanging at:

`[27/05/2025 12:03:43.000] [WARN] ConfigManager locked, waiting...
`
With that other setup, I had to reboot the host to recover.

Work around: resize SDR++ before connecting to the SDR, don't resize it afterwads.

At some point, sometime stopping/starting SDR++ decoding, after hours, SDR++ hangs also, with the same symptoms, a freeze, then no way to kill/restart SDR++.

That SDR++ app was buill using git clone and the latest source codes. After the crash then kill of SDR++, SDR++ hangs at "Loading config":

```
bkant@trixie-gl:~/SDRPlusPlus/build$ ipcs -a

------ Message Queues --------
key        msqid      owner      perms      used-bytes   messages    

------ Shared Memory Segments --------
key        shmid      owner      perms      bytes      nattch     status      
0x00000000 11         bkant      600        524288     2          dest         
0x00000000 14         bkant      600        524288     2          dest         
0x00000000 327695     bkant      600        524288     2          dest         
0x00000000 18         bkant      600        524288     2          dest         
0x00000000 23         bkant      600        524288     2          dest         
0x00000000 28         bkant      600        33554432   2          dest         
0x00000000 229405     bkant      600        67108864   2          dest         
0x00000000 327711     bkant      600        524288     2          dest         
0x00000000 35         bkant      600        524288     2          dest         
0x00000000 38         bkant      600        524288     2          dest         
0x00000000 42         bkant      600        524288     2          dest         
0x00000000 51         bkant      600        1048576    2          dest         
0x00000000 131132     lightdm    600        524288     2          dest         
0x00000000 131133     lightdm    600        33554432   2          dest         

------ Semaphore Arrays --------
key        semid      owner      perms      nsems     

bkant@trixie-gl:~/SDRPlusPlus/build$ ls /dev/shm/
'Glbl\sdrSrvCmdSema'  'Glbl\sdrSrvComMtx'  'Glbl\sdrSrvComShMem'  'Glbl\sdrSrvRespSema'
bkant@trixie-gl:~/SDRPlusPlus/build$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU Tablet
Bus 001 Device 003: ID 1df7:3060 SDRplay RSPdxR2
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
bkant@trixie-gl:~/SDRPlusPlus/build$ ../../vglrun ./sdrpp -r ../root_dev/
[30/05/2025 21:35:57.000] [INFO] SDR++ v1.2.1
[30/05/2025 21:35:57.000] [INFO] Loading config
```
 
- CPU: AMD Ryzen 9 7945HX
- RAM: 96GB
- GPU: that of the Ryzen plus emulation by CPU, using Linux KVM/QEMU and VirtualGL
- SDR: local, via USB, the SDRplay RSPdxR2 being attached to the VM

[image]

AlexandreRouma, 2025-05-30:
I already told you I don't support running SDR++ in a VM, stop wasting my time with this.

closed 2025-05-30 by AlexandreRouma

bk4nt, 2025-05-30:
Hi, I was very clear in my title, allowing you to quickly close this case. My only point was: what could prevent SDR++ to start after such a crash, what might remain locked.

And sorry, I didn't find any forum to discuss such topics.

bk4nt, 2025-06-19:
Hi,

I added a GPU to my setup for other needs like AI or ffmpeg.

By the way, I'm now able to use sdr++ and my Sdrplay device inside a Windows VM, via the network, using a RDP session. That being supported by AMD/Intel, Linux, Microsoft and Nvidia.

I'm so noticing a GPU for GL is offloading a lot the CPU, and the sound out of this VM via RDP is very good. I used also sdr++ on a Windows box, via RDP but without an adequate GPU; that was loading a lot the CPU and I was getting many audio drops...

I'm very happy so also, using Windows and a Nvidia GPU. But the costs and the hardware build aren't the same. Maybe this will change later, when Linux envs and tools for GL using KVM/VMs and GPU sharing will be less experimental.

[image]