AlexandreRouma/SDRPlusPlus #1650: Android menu hidden

LinuxLover9, 2025-06-22:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Pixel7
- RAM: Pixel7
- GPU: Pixel7
- SDR: (Remote or local? If remote, what protocol?) Rtl-tcp

**Software**
- Operating System: grapheneOS Android 15
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software). 1.0.6

**Bug Description**
A clear description of the bug.
Only half of the menu is visible. I cannot change the DPI setting.

**Steps To Reproduce**
1. Load s/w
2. Select menu display high dpi
3. No option to change the setting.

[image]

[image]

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

KubaPro010, 2025-06-22:
no big here, just drag the menu border right and that's the fix

LinuxLover9, 2025-07-16:
OK, thanks.