AlexandreRouma/SDRPlusPlus #1655: Audio sink underrun using libresdr at high (20M) sample rate

gariac, 2025-07-06:
I had to ponder which issue section to post this. 

Details:
`
Running v.1.2.1 compiled from source with -j2 option.
Operating System: Debian GNU/Linux 13
KDE Plasma Version: 6.3.5
KDE Frameworks Version: 6.13.0
Qt Version: 6.8.2
Kernel Version: 6.12.33+deb13-amd64 (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 7840U w/ Radeon  780M Graphics
Memory: 90.2 GiB of RAM
Graphics Processor: AMD Radeon 780M
Manufacturer: Framework
Product Name: Laptop 13 (AMD Ryzen 7040Series)
System Version: A7
`

Thanks to /F5OEO/tezuka_fw, I can do very high sample rates using ethernet. I decided to see at what sample rate I could break the program. At 15M, everything works. Around 18M I could hear obvious distortion but no error or warning message. You can see I got underrun warning at 21M. I picked a classical radio station so I am using the WFM demod.  I plan on repeating the test using the weather radio once I can get a solid signal. It is the only steady NFM source.

 https://imgur.com/a/WSZcJKF

`
[06/07/2025 13:57:31.000] [INFO] New DSP samplerate: 23500000.000000 (source samplerate is 23500000.000000)
[06/07/2025 13:57:40.000] [INFO] PlutoSDRSourceModule 'PlutoSDR Source': Start!
[06/07/2025 13:57:40.000] [INFO] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 104900000.000000!
[06/07/2025 13:57:40.000] [WARN] AudioSinkModule Warning: RtApiAlsa::callbackEvent: audio write error, underrun. (1)
[06/07/2025 13:57:49.000] [WARN] AudioSinkModule Warning: RtApiAlsa::callbackEvent: audio write error, underrun. (1)
[06/07/2025 13:57:55.000] [WARN] AudioSinkModule Warning: RtApiAlsa::callbackEvent: audio write error, underrun. (1)
[06/07/2025 13:58:08.000] [INFO] PlutoSDRSourceModule 'PlutoSDR Source': Stop!
[Resamp] predec: 64, interp: 16, decim: 21, inacc: 0.000000%, taps: 1596
[06/07/2025 13:58:17.000] [INFO] New DSP samplerate: 21000000.000000 (source samplerate is 21000000.000000)
[06/07/2025 13:58:25.000] [INFO] PlutoSDRSourceModule 'PlutoSDR Source': Start!
[06/07/2025 13:58:25.000] [INFO] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 104900000.000000!
[06/07/2025 13:58:42.000] [WARN] AudioSinkModule Warning: RtApiAlsa::callbackEvent: audio write error, underrun. (1)
[06/07/2025 13:59:00.000] [INFO] PlutoSDRSourceModule 'PlutoSDR Source': Stop!
[Resamp] predec: 64, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[06/07/2025 13:59:12.000] [INFO] New DSP samplerate: 20000000.000000 (source samplerate is 20000000.000000)
[06/07/2025 13:59:20.000] [INFO] PlutoSDRSourceModule 'PlutoSDR Source': Start!
[06/07/2025 13:59:20.000] [INFO] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 104900000.000000!

`
I kept increasing the -j compilation option but it isn't a matter of the number of CPUs.  I left it at -j2. 06/07/2025 14:16:19.000] [INFO] New DSP samplerate: 15000000.000000 (source samplerate is 15000000.000000)
[06/07/2025 14:16:27.000] [INFO] PlutoSDRSourceModule 'PlutoSDR Source': Start!
[06/07/2025 14:16:27.000] [INFO] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 104900000.000000!
[06/07/2025 14:17:03.000] [WARN] AudioSinkModule Warning: RtApiAlsa::callbackEvent: audio write error, underrun. (1)

Going back to the 15M sample rate, I can cause the underrun by increasing the FFT resolution to  524288. The CPUs are not taxed per btop.

`
06/07/2025 14:16:19.000] [INFO] New DSP samplerate: 15000000.000000 (source samplerate is 15000000.000000)
[06/07/2025 14:16:27.000] [INFO] PlutoSDRSourceModule 'PlutoSDR Source': Start!
[06/07/2025 14:16:27.000] [INFO] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 104900000.000000!
[06/07/2025 14:17:03.000] [WARN] AudioSinkModule Warning: RtApiAlsa::callbackEvent: audio write error, underrun. (1)

`
Here is the btop output under this condition:
https://imgur.com/a/RijT3l3

AlexandreRouma, 2025-07-06:
Not an SDR++ issue

closed 2025-07-06 by AlexandreRouma

gariac, 2025-07-07:
That was why I was wondering under which category to put it. It seems to me that it is a sdrpp issue for the obvious (to me) reason. The data rate fed to the sound device is independent of the sample rate of the SDR.  The sound process has no CPU issue under this failing condition.  My next experiment is to get my usb DAC and feed it some audio from  audacious while sdrpp is streaming to the built in sound. If the audacious audio is clear then it is a sdrpp issue.

 Also the FFT size effect happens at even USB speeds though it isn't a big deal. I discovered this when the dropped data problem went away when I used the zoom feature on the spectrum width.

gariac, 2025-07-08:
Doing some hacking, I can make the underflow messages go away.  Please leave the issue open for a while  since I may stumble on a fix.

Current settings:
`
pw-metadata -n settings
Found "settings" metadata 32
update: id:0 key:'log.level' value:'2' type:''
update: id:0 key:'clock.rate' value:'48000' type:''
update: id:0 key:'clock.allowed-rates' value:'[ 48000 ]' type:''
update: id:0 key:'clock.quantum' value:'1024' type:''
update: id:0 key:'clock.min-quantum' value:'2048' type:''
update: id:0 key:'clock.max-quantum' value:'8192' type:''
update: id:0 key:'clock.force-quantum' value:'0' type:''
update: id:0 key:'clock.force-rate' value:'0' type:''

`
Original settings:
`
pw-metadata -n settings
Found "settings" metadata 32
update: id:0 key:'log.level' value:'2' type:''
update: id:0 key:'clock.rate' value:'48000' type:''
update: id:0 key:'clock.allowed-rates' value:'[ 48000 ]' type:''
update: id:0 key:'clock.quantum' value:'1024' type:''
update: id:0 key:'clock.min-quantum' value:'32' type:''
update: id:0 key:'clock.max-quantum' value:'2048' type:''
update: id:0 key:'clock.force-quantum' value:'0' type:''
update: id:0 key:'clock.force-rate' value:'0' type:''
`

gariac, 2025-07-08:
I found this old sdrpp issue:
https://github.com/AlexandreRouma/SDRPlusPlus/issues/131

The tldr is the recording sounds good but the playback skips. I disagree.

15Msps:
https://voca.ro/1nrqFaP8pGDw

22Msps:
https://voca.ro/11NmeFjaFaKf

The next experiment is to just save IQ so that sdrpp isn't even using the alsa plugin. This require a demod program. I'm thinking adsb . If the error rate looks similar, then we know sdrpp is at least producing a clean baseband signal independent of sample rate.