AlexandreRouma/SDRPlusPlus #1657: Android - RTL-SDR Blog 4

NexGen-3D-Printing, 2025-07-07:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Intel
- RAM: 8GB
- GPU: Intel
- SDR: Local

**Software**
- Operating System: Android 13 (x86 BlissOS)
- SDR++: 1.2.1 + June 24 2025

**Bug Description**
Can't see the Blog 4 at all, RTL-SDR is selected, refresh does nothing, drop down box is empty, the dongle does work with SDR Touch and SDRangel and a couple of other apps I tried, each time I start the app is ask for permission to connect to different devices, like my GNSS device and my LTE adaptor, and sometimes the Blog 4, but it never shows up in the list, I have tried removing all devices and the app still cant see the dongle, I have tested directly connecting it to a USB-C port, and have tried through a powered USB-C hub, but nothing has worked, the device being used is Dell Latitude Detachable with BlissOS, Android 13, everything works on this device as expected, have tested with and without the SDR Touch driver installed.

**Steps To Reproduce**
1. Install App
2. Run App
3. Select SDR-RTL then hit refresh, and nothing appears in the second drop down list.

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

NexGen-3D-Printing, 2025-07-07:
Additionally, I just tested a Nooelec NesDR Smart USB Dongle, and this also does not detect and work under SDR++, but the dongle does work on all other Apps tested, including MagicSDR.

AlexandreRouma, 2025-08-07:
the blog v4 works fine on android. Make sure you DO NOT have the driver app installed (it hoards control of the dongle for any app that has its own driver like SDR++) or use the RTL TCP source if you want to use the driver app.

closed 2025-08-07 by AlexandreRouma