AlexandreRouma/SDRPlusPlus #1660: Request to compile apk for earlier version.

agass1, 2025-07-09:
**Feature description**
Brother im sorry for this ask.
I want ask you to compile build for android oreo.I have a tablet not bad,but it couldn't be updated with any software greater than android O 8.0.0.
I cannot use custom ROM,because huawei agassi ags-w09 isnt supported.
I only need this app on it.
Will be glad if you will help me with.
I swear multiple people have same problem,and they will be very respectful for build that can be even experenentally launched on such devices.
More than i tried to patch APK.
Even more patched APK working great on new devices but on android 8 i could only install,launch fails.
If you wouldn't work with this problem please tell me how to install these project properly in android studio,or just share preconfigured folder with this project for android studio.
Thank you very much,i hope anyway you will help.

agass1, 2025-07-09:
Add:
If i understood right,difficulties to build for SDK 26 is:
-OpenGL ES3 to OpenGL ES2.
-Library problems.
Right?

AlexandreRouma, 2025-07-09:
Would require way too many compromises to work properly. Not planned.

closed 2025-07-09 by AlexandreRouma