AlexandreRouma/SDRPlusPlus #1681: Add MIDI controller support

Hagtronics, 2025-08-27:
This may well be "out of scope" for you, and if it is please feel free to delete this post.  :-)

One really nice feature of piHPSDR is the possibility to use a commercial / custom MIDI keyboard to control the various controls on the software. The dl1ycf fork of piHPSDR went even further by adding improved GUI wrappers around the various functions. MIDI itself is pretty easy in C++ with many libraries available. I got RtMidi going in just an hour on a Raspberry Pi 4, Windows is easy also. I really don't see this applying to Android. 

The hard(er) part is making the connection / control matrix so the user can assign MIDI functions to various controls in SDR++.

Thanks for your hard work.

dh1tw, 2025-10-04:
I just did a quick prototype and added a Plugin that allows MIDI devices to control SDR++. I really like it, it improves the useability of SDR++ for practical ham usage. Apart of tuning the VFO and the center frequency I find it particularly useful to have Waterfall Min, Max and Zoom on rotary knobs for quick adjustment. 

Unfortunately, implementing this plugin properly, would require some modifications of the SDR++ source code. Some of the variables (e.g. the ones that hold the values of the sliders) are actually private and not exposed. In case Alexandre would accept a few changes, I could create a PR which addes Setter and Getter Methods for some needed Control variables.

bubnikv, 2026-04-16:
@dh1tw Thobias, what MIDI hardware do you use?
73, Vojtech OK1IAK