AlexandreRouma/SDRPlusPlus #1682: 100% GPU usage even when doing absolutely nothing

whoreson, 2025-08-30:
**Hardware**
- CPU: i5
- RAM: 32GB
- GPU: Radeon R7
- SDR: ---
-
**Software**
- Operating System: Windows 7 SP1
- SDR++: 1.2.1 Aug 26 2025

**Bug Description**
GPU-Z shows the GPU working at full capacity doing nothing but generating heat.

**Steps To Reproduce**
1. Install and start GPU-Z (or whatever monitoring software I guess)
2. Start SDRpp

Hagtronics, 2025-08-30:
Did you use the pre-compiled binaries from Github or build them from source yourself?

I got the latest nightly pre-compiled binaries and ran it on my Win 7 SP1 computer. No issues. My computer uses a:  AMD FirePro M4170 (1 GB dedicated GDDR5) graphics card however. My overall CPU usage for SDR++ was like 2% Idle, and 20% running.

It might be more useful to see what actual process is taking all the CPU time, or Memory - you can do this via the Windows Task Manager.

Hope this helps.

whoreson, 2025-08-30:
Oh really.

Well, then, quite the festive event here: this very ticket marks the _very_ last time I bothered to report an issue/bug.

KubaPro010, 2025-08-30:
This is a real issue, as my PC with a GTX 1080 Ti, with a browser, discord, voicemeeter and other stuff only has a usage of 2%, but when SDR++ comes around, it goes up to 23% while doing nothing, simple imgui should not take 20% of a GPU with the performance of the RTX 2080

AlexandreRouma, 2025-08-30:
Run the ImGui example (with the same ImGui version and backend) and see.
There's nothing I can do on my end besides drop ImGui which is being done for 2.0

bubnikv, 2025-08-31:
> It's an immediate mode gui which redraws on every frame no matter if something changed, of course it's gonna use more resources.

We are using ImGUI in PrusaSlicer, where we perform screen refresh only if we internally mark the screen as changed. It is a bit involved though, because some ImGUI state modifications require two, some even more screen refreshes. If I remember correctly, most of the UI interactions require two screen updates (keyboard input, scroll bars...), while resizing of boxes required more refreshes.

At the end of the day implementing the state machine to reduce screen updates was not that difficult and it improved application performance a lot. Certainly worthy.

whoreson, 2025-08-31:
BTW starting SDRpp is a 0->14% GPU usage jump on Intel, and the same 100% GPU usage on nVidia as well as on AMD.

bubnikv, 2025-08-31:
I don't know your systems. I suppose IMGUI does a screen refresh on VSYNC, so the CPU load will depend on screen refresh rate, possible variable screen refresh rate, screen resolution etc. Too many factors to judge your particular case from your description.

AlexandreRouma, 2025-08-31:
> At the end of the day implementing the state machine to reduce screen updates was not that difficult and it improved application performance a lot. Certainly worthy.

It's just polishing a turd. ImGUI was not designed for that, as you've found out yourself by how it sometimes requires multiple screen updates.

Instead I have *already* written a custom GUI library for SDR++ 2.0 which is event based and uses exactly 0% CPU and 0% GPU when nothing changes all while making SDR++ Server a lot more flexible.

[image]

whoreson, 2025-11-28:
> Any update on this?

all is not lost, i've converted 'natpos' (https://www.ecstaticlyrics.com/radio/sdr/natpos/) from GLFW to plain X11, and will do a native Windows port as well. then various audio outputs, and we'll have a nice portable SDR software that doesn't murder your system for a few sliders